project-chip/connectedhomeip #35663: [1.2] I can't add device to chip-tool

yiwang103, 2024-09-19:
### Reproduction steps

I'm maybe first add device to chip-tool is success, and not delete device, but factory reset device. Then second add device to chip-tool is failed, and it has been a failure. I try factory reset device or gateway, still no use. But change gateway occasionally can success.

### Bug prevalence

if appear, on once gateway, always

### GitHub hash of the SDK that was being used

181b0cb14ff007ec912f2ba6627e05dfb066c008

### Platform

freeRTOS

### Type

Core SDK Interopability Issue

### Anything else?

[1726197597.300182][45989:46000] CHIP:CTL: Setting thread operational dataset from parameters
[1726197597.300264][45989:46000] CHIP:CTL: Setting attestation nonce to random value
[1726197597.300395][45989:46000] CHIP:CTL: Setting CSR nonce to random value
[1726197597.300544][45989:46000] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1726197597.300589][45989:46000] CHIP:CTL: Starting commissioning discovery over BLE
[1726197597.300640][45989:46000] CHIP:CTL: Starting commissioning discovery over DNS-SD
[2024-09-13 11:19:57.325] [info] topic: redbox-middlerware/report
[2024-09-13 11:19:57.325] [info] payload: {"module":"redbox-middlerware","method":"reply","timestamp":1726197597286,"commandId":5151,"version":1,"manfName":"redbox","payload":{"type":"thread","open_network":"","status":0}}
[2024-09-13 11:19:57.325] [info] web topic response or report
[1726197597.340570][45989:45999] CHIP:BLE: BLE removing known devices.
[1726197597.360335][45989:45999] CHIP:BLE: BLE initiating scan.
[1726197597.974667][45989:46000] CHIP:DL: Long dispatch time: 673 ms, for event type 2
[1726197597.975172][45989:46000] CHIP:EM: Retransmitting MessageCounter:154649969 on exchange 57720i Send Cnt 2
[1726197598.019010][45989:46000] CHIP:DMG: OnResubscribeTimerCallback: ForceCASE = 1
[1726197598.019212][45989:46000] CHIP:DMG: Trying to establish a CASE session for subscription
[1726197598.019264][45989:46000] CHIP:CSM: FindOrEstablishSession: PeerId = [1:0000000000000001]
[1726197598.019323][45989:46000] CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1726197598.019474][45989:46000] CHIP:DIS: OperationalSessionSetup[1:0000000000000001]: State change 1 --> 2
[1726197598.024548][45989:46000] CHIP:DIS: Lookup clearing interface for non LL address
[1726197598.024792][45989:46000] CHIP:DIS: UDP:[fd7b:e51c:20f5:1:eab4:3de4:c014:c6e7%eth0]:5540: new best score: 5
[1726197598.024901][45989:46000] CHIP:DIS: Checking node lookup status after 5 ms
[1726197598.024932][45989:46000] CHIP:DIS: Keeping DNSSD lookup active
[1726197598.026541][45989:46000] CHIP:DIS: UDP:[fd7b:e51c:20f5:1:eab4:3de4:c014:c6e7%eth0]:5540: score has not improved: 5
[1726197598.026654][45989:46000] CHIP:DIS: Checking node lookup status after 7 ms
[1726197598.026673][45989:46000] CHIP:DIS: Keeping DNSSD lookup active
[1726197598.089368][45989:45999] CHIP:BLE: Device 6D:42:CF:95:9F:CB does not look like a CHIP device.
[1726197598.175727][45989:45999] CHIP:BLE: New device scanned: E9:91:0E:20:AD:8B
[1726197598.175867][45989:45999] CHIP:BLE: Device discriminator match. Attempting to connect.
[1726197598.220633][45989:46000] CHIP:DIS: Checking node lookup status after 201 ms
[1726197598.220901][45989:46000] CHIP:DIS: OperationalSessionSetup[1:0000000000000001]: Updating device address to UDP:[fd7b:e51c:20f5:1:eab4:3de4:c014:c6e7]:5540 while in state 2
[1726197598.221044][45989:46000] CHIP:DIS: OperationalSessionSetup[1:0000000000000001]: State change 2 --> 3
[1726197598.221571][45989:46000] CHIP:IN: SecureSession[0x7f7c0db310]: Allocated Type:2 LSID:41948
[1726197598.221813][45989:46000] CHIP:SC: Initiating session on local FabricIndex 1 from 0x000000000001B669 -> 0x0000000000000001
[1726197598.237151][45989:46000] CHIP:EM: <<< [E:57721i S:0 M:154649970] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[fd7b:e51c:20f5:1:eab4:3de4:c014:c6e7]:5540] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
[1726197598.237708][45989:46000] CHIP:SC: Sent Sigma1 msg
[1726197598.237771][45989:46000] CHIP:DIS: OperationalSessionSetup[1:0000000000000001]: State change 3 --> 4
[1726197599.173903][45989:46000] CHIP:EM: Retransmitting MessageCounter:154649970 on exchange 57721i Send Cnt 1
[1726197599.258589][45989:45999] CHIP:DL: ConnectDevice complete
[1726197599.509173][45989:46000] CHIP:EM: Retransmitting MessageCounter:154649969 on exchange 57720i Send Cnt 3
[1726197599.564506][45989:45999] CHIP:DL: Char /org/bluez/hci0/dev_E9_91_0E_20_AD_8B/service000a on service /org/bluez/hci0/dev_E9_91_0E_20_AD_8B/service000a
[1726197599.564912][45989:45999] CHIP:DL: Char /org/bluez/hci0/dev_E9_91_0E_20_AD_8B/service0006 on service /org/bluez/hci0/dev_E9_91_0E_20_AD_8B/service000a
[1726197599.564948][45989:45999] CHIP:DL: Char /org/bluez/hci0/dev_E9_91_0E_20_AD_8B/service0006 on service /org/bluez/hci0/dev_E9_91_0E_20_AD_8B/service000a
[1726197599.565016][45989:45999] CHIP:DL: Char /org/bluez/hci0/dev_E9_91_0E_20_AD_8B/service000a on service /org/bluez/hci0/dev_E9_91_0E_20_AD_8B/service000a
[1726197599.565039][45989:45999] CHIP:DL: Char /org/bluez/hci0/dev_E9_91_0E_20_AD_8B/service000a on service /org/bluez/hci0/dev_E9_91_0E_20_AD_8B/service000a
[1726197599.565083][45989:45999] CHIP:DL: New BLE connection: conn 0x7f7806fc20, device E9:91:0E:20:AD:8B, path /org/bluez/hci0/dev_E9_91_0E_20_AD_8B
[1726197599.565315][45989:46000] CHIP:DL: HandlePlatformSpecificBLEEvent 16385
[1726197599.565352][45989:46000] CHIP:CTL: Discovered device to be commissioned over BLE
[1726197599.565411][45989:46000] CHIP:CTL: Attempting PASE connection to BLE
[... 74 lines cut ...]
[1726197611.506686][45989:46000] CHIP:IN: SecureSession[0x7f7c0db310]: Released - Type:2 LSID:41948
[1726197611.506885][45989:46000] CHIP:DIS: OperationalSessionSetup[1:0000000000000001]: State change 4 --> 2
[1726197611.507014][45989:46000] CHIP:DIS: OperationalSessionSetup[1:0000000000000001]: State change 2 --> 1
[1726197611.507166][45989:46000] CHIP:DMG: Failed to establish CASE for re-subscription with error 'src/protocols/secure_channel/CASESession.cpp:553: CHIP Error 0x00000032: Timeout'
[1726197611.507243][45989:46000] CHIP:DMG: MoveToState ReadClient[0x7f7c0403f0]: Moving to [      Idle]
[1726197611.507491][45989:46000] CHIP:DMG: Will try to resubscribe to 01:0000000000000001 at retry index 7 after 70074ms due to error src/protocols/secure_channel/CASESession.cpp:553: CHIP Error 0x00000032: Timeout
[1726197611.507591][45989:46000] CHIP:IN: SecureSession[0x7f7c016a30]: MarkAsDefunct Type:2 LSID:41930
[1726197613.296094][45989:46000] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1726197615.769870][45989:45999] CHIP:DL: Indication received, conn = 0x7f7806fc20
[1726197615.770271][45989:46000] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
^[^[[24~^[a[1726197617.975512][45989:46000] CHIP:CTL: Commissioning discovery over BLE failed: src/platform/Linux/bluez/ChipDeviceScanner.cpp:173: CHIP Error 0x00000032: Timeout
[1726197617.975724][45989:46000] CHIP:-: src/platform/Linux/bluez/ChipDeviceScanner.cpp:173: CHIP Error 0x00000032: Timeout at src/controller/SetUpCodePairer.cpp:324
[1726197617.975794][45989:46000] CHIP:BLE: BLE scan error: src/platform/Linux/bluez/ChipDeviceScanner.cpp:173: CHIP Error 0x00000032: Timeout
[1726197618.336466][45989:46000] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
A[1726197620.812730][45989:45999] CHIP:DL: Indication received, conn = 0x7f7806fc20
[1726197620.813376][45989:46000] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1726197623.376389][45989:46000] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1726197625.852422][45989:45999] CHIP:DL: Indication received, conn = 0x7f7806fc20
[1726197625.852914][45989:46000] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1726197627.301985][45989:46000] CHIP:CTL: Discovery timed out
[1726197627.302112][45989:46000] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1726197627.302135][45989:46000] CHIP:TOO: Secure Pairing Failed
[1726197627.302147][45989:46000] CHIP:TOO: LogJoinFail
[1726197627.302350][45989:46000] CHIP:TOO: Command: {"join":{"error":1}}
[2024-09-13 11:20:27.302] [info] error: 1

bzbarsky-apple, 2024-09-19:
When you commission the second time, are you trying to use the same node ID as before?  If so, you might be running into cached DNS-SD advertisement issues....

andy31415, 2024-10-21:
If thread is used, the border router will even refuse to use the same nodeid for advertisement.

@yiwang103 please provide a reproduction script with a test application for this for us to understand. The logs themselves are not sufficient for ease of debugging.