project-chip/connectedhomeip #35665: [CERT-TEST-FAILURE] TC-CNET-4.11 step7

tianfeng-yang, 2024-09-19:
### Feature Area

Area:ConcentrationMeasurement, Other

### Test Case

TC-CNET-4.11

### Reproduction steps

In `step 7` of `TC-CNET-4.11`, when connecting to SSID2, according to the current processing logic, if the network where TH is located is not in the same LAN as SSID2, it will definitely not receive a response. Does this processing logic meet expectations?

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

5bff3a5a4ea940801491d7ca120d2bf0be77c19e

### Platform

raspi

andy31415, 2024-10-21:
Removed linux tag: this seems to be a cert failure rather than platform specific.

Jim-tech, 2024-10-23:
This issue is also blocking our certification as well. In the spec 11.8.7.9, it says:
[image]
In our product, the ConnectMaxTimeSeconds is set to 20 seconds. 
Apparently, the chip-tool doesn't wait enough time before it determines to timeout.

tianfeng-yang, 2024-10-23:
> Removed linux tag: this seems to be a cert failure rather than platform specific.

Yes, it has nothing to do with the Linux platform. The purpose of this test case is to switch from network 1 to network 2. After network 1 receives the command, according to the current logic, the device immediately starts the switching logic and performs a callback in network 2. At this time, TH cannot get a response on network 1.