project-chip/connectedhomeip #35670: [CERT-TEST-FAILURE] [TC-IDM-10.3] Several ClusterRevision values are not in sync with the spec

bauerschwan, 2024-09-19:
### Feature Area

Other

### Test Case

TC-IDM-10.3

### Reproduction steps

When running TC-IDM-10.3 the following errors occur:
`
[MatterTest] 09-19 10:54:04.243 INFO
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.3
    location:
       Endpoint: 0,
       Cluster:  29 (0x1d) Descriptor
      Attribute:65533 (0xfffd)
    problem: Revision found on cluster (2) does not match revision listed in the spec (3)
    spec_location:

[MatterTest] 09-19 10:54:04.243 INFO
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.3
    location:
       Endpoint: 0,
       Cluster:  31 (0x1f) AccessControl
      Attribute:65533 (0xfffd)
    problem: Revision found on cluster (1) does not match revision listed in the spec (2)
    spec_location:

[MatterTest] 09-19 10:54:04.243 INFO
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.3
    location:
       Endpoint: 0,
       Cluster:  49 (0x31) NetworkCommissioning
      Attribute:65533 (0xfffd)
    problem: Revision found on cluster (2) does not match revision listed in the spec (3)
    spec_location:
`

These clusters have hard-coded ClusterRevision in the code an d are not yet updated to the spec.

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

17b1a38e909e7874593bcb87c31be03a5866f1d4

### Platform

nrf

bauerschwan, 2024-09-19:
See also https://github.com/project-chip/connectedhomeip/issues/34227

cecille, 2024-09-19:
Please see https://github.com/project-chip/connectedhomeip/pull/35607 - spec changes for 0.9 landed after the SVE cut, so the data model files didn't get updated. The in-progress ones should work, you will need to pull this change because the TH doesn't have an override.

cjandhyala, 2024-10-02:
closing the issue as the PR is merged. if you think this issue is still open, pls re-open it @bauerschwan

closed 2024-10-02 by cjandhyala