project-chip/connectedhomeip #35709: [BUG] src/access/tests/TestAccessControl.cpp Fails Under G++ 8.2.0 with 'const chip::Access::CheckData [57]{ ... }' Is Not a Constant Expression

gerickson, 2024-09-21:
### Reproduction steps

When configuring the Matter build with:

```
Creating ".build/Target.mak/${BUILD_PRODUCT}/digi/dey/8.2.0/release"
mkdir -p ".build/Target.mak/${BUILD_PRODUCT}/digi/dey/8.2.0/release"
Creating ".build/Target.mak/${BUILD_PRODUCT}/digi/dey/8.2.0/release/build.ninja"
PKG_CONFIG_PATH="${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/avahi/usr/lib/pkgconfig:${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/dbus/usr/lib/pkgconfig:${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/glib/usr/lib/pkgconfig:${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/openssl/usr/lib/pkgconfig" \
            ${BUILD_ROOT}/third_party/matter/repo/.environment/cipd/packages/pigweed/gn \
    --threads=1 \
    --root=${BUILD_ROOT}/third_party/matter/repo \
    gen \
    --args='custom_toolchain="custom" target_ar="${SYSROOT}/usr/bin/arm-dey-linux-gnueabi/arm-dey-linux-gnueabi-ar" target_cc="${SYSROOT}/usr/bin/arm-dey-linux-gnueabi/arm-dey-linux-gnueabi-gcc" target_cxx="${SYSROOT}/usr/bin/arm-dey-linux-gnueabi/arm-dey-linux-gnueabi-g++" target_cflags_c=["--sysroot=${BUILD_ROOT}/rootfs","-mcpu=cortex-a8","-mfloat-abi=hard","-mfpu=neon","-fno-omit-frame-pointer","-fno-strict-aliasing","-I${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/avahi/usr/include","-I${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/glib/usr/include/gio-unix-2.0","-I${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/glib/usr/include/glib-2.0","-I${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/glib/usr/lib/glib-2.0/include","-I${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/openssl/usr/include"] target_cflags_cc=["--sysroot=${BUILD_ROOT}/rootfs","-mcpu=cortex-a8","-mfloat-abi=hard","-mfpu=neon","-fno-omit-frame-pointer","-fno-strict-aliasing","-I${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/avahi/usr/include","-I${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/glib/usr/include/gio-unix-2.0","-I${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/glib/usr/include/glib-2.0","-I${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/glib/usr/lib/glib-2.0/include","-I${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/openssl/usr/include"] target_ldflags=["--sysroot=${BUILD_ROOT}/rootfs","-L${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/avahi/usr/lib/","-L${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/avahi/usr/lib/","-L${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/avahi/usr/lib/","-L${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/avahi/usr/lib/","-L${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/openssl/usr/lib/","-L${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/openssl/usr/lib/","-L${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/glib/usr/lib/","-lgio-2.0","-L${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/glib/usr/lib/","-lglib-2.0","-L${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/glib/usr/lib/","-lgobject-2.0","-L${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/glib/usr/lib/","-lgmodule-2.0","-L${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/pcre/usr/lib/","-lpcre","-Wl,-rpath-link","-Wl,${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/dbus/usr/lib","-Wl,-rpath-link","-Wl,${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/zlib/usr/lib","-Wl,-rpath-link","-Wl,${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/libcap/lib","-Wl,-rpath-link","-Wl,${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/expat/usr/lib","-Wl,-rpath-link","-Wl,${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/util-linux/lib","-Wl,-rpath-link","-Wl,${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/libffi/usr/lib"] target_os="linux" target_cpu="arm" arm_cpu="cortex-a8" chip_crypto="openssl" chip_enable_ble=true chip_enable_openthread=false chip_enable_wifi=true chip_mdns="platform" chip_target_is_big_endian=false chip_target_style="unix" chip_with_lwip=false enable_exceptions=false enable_rtti=false is_debug=true sysroot="${BUILD_ROOT}/rootfs"' \
    .build/Target.mak/${BUILD_PRODUCT}/digi/dey/8.2.0/release \
```

and then building it with:

```
Processing ".build/Target.mak/${BUILD_PRODUCT}/digi/dey/8.2.0/release/build.ninja"
ZAP_INSTALL_PATH=${BUILD_ROOT}/third_party/matter/repo/.environment/cipd/packages/zap/ \
    ninja \
    --verbose \
    -j 1 \
    -C .build/Target.mak/${BUILD_PRODUCT}/digi/dey/8.2.0/release
ninja: Entering directory `.build/Target.mak/${BUILD_PRODUCT}/digi/dey/8.2.0/release'
...
[301/2441] ${SYSROOT}/usr/bin/arm-dey-linux-gnueabi/arm-dey-linux-gnueabi-g++ -MMD -MF obj/src/access/tests/TestAccessControl.lib.TestAccessControl.cpp.o.d -Wconversion -mcpu=cortex-a8 -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wundef -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color --sysroot=${BUILD_ROOT}/rootfs -fno-strict-aliasing -fmacro-prefix-map=../../../../../../../repo/= -D_REENTRANT -std=gnu++17 -fno-rtti -Wnon-virtual-dtor --sysroot=${BUILD_ROOT}/rootfs -mcpu=cortex-a8 -mfloat-abi=hard -mfpu=neon -fno-omit-frame-pointer -fno-strict-aliasing -I${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/avahi/usr/include -I${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/glib/usr/include/gio-unix-2.0 -I${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/glib/usr/include/glib-2.0 -I${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/glib/usr/lib/glib-2.0/include -I${BUILD_ROOT}/results/${BUILD_PRODUCT}/digi/dey/8.2.0/release/third_party/openssl/usr/include -DCHIP_HAVE_CONFIG_H=1 -DPW_SPAN_ENABLE_ASSERTS=false -DPW_UNIT_TEST_CONFIG_MEMORY_POOL_SIZE=16384 -I../../../../../../../repo/src/include -I../../../../../../../repo/src -Igen/include -I../../../../../../../repo/config/standalone -I../../../../../../../repo/zzz_generated/app-common -I../../../../../../../repo/third_party/nlassert/repo/include -I../../../../../../../repo/third_party/nlio/repo/include -I../../../../../../../repo/third_party/nlfaultinjection/include -I../../../../../../../repo/third_party/inipp/repo/inipp -I../../../../../../../repo/third_party/pigweed/repo/pw_string/public -I../../../../../../../repo/third_party/pigweed/repo/pw_assert/public -I../../../../../../../repo/third_party/pigweed/repo/pw_assert/assert_compatibility_public_overrides -I../../../../../../../repo/third_party/pigweed/repo/pw_preprocessor/public -I../../../../../../../repo/third_party/pigweed/repo/pw_polyfill/public -I../../../../../../../repo/third_party/pigweed/repo/pw_assert_log/check_backend_public_overrides -I../../../../../../../repo/third_party/pigweed/repo/pw_assert_log/public -I../../../../../../../repo/third_party/pigweed/repo/pw_log/public -I../../../../../../../repo/third_party/pigweed/repo/pw_log_basic/public_overrides -I../../../../../../../repo/third_party/pigweed/repo/pw_log_basic/public -I../../../../../../../repo/third_party/pigweed/repo/pw_span/public -I../../../../../../../repo/third_party/pigweed/repo/pw_status/public -I../../../../../../../repo/third_party/pigweed/repo/pw_result/public -I../../../../../../../repo/third_party/pigweed/repo/third_party/fuchsia/repo/sdk/lib/stdcompat/include -I../../../../../../../repo/third_party/pigweed/repo/pw_unit_test/public -I../../../../../../../repo/third_party/pigweed/repo/pw_unit_test/light_public_overrides -I../../../../../../../repo/third_party/pigweed/repo/pw_unit_test/public_overrides -c ../../../../../../../repo/src/access/tests/TestAccessControl.cpp -o obj/src/access/tests/TestAccessControl.lib.TestAccessControl.cpp.o
```

Fails with:

```
[... cut ...]
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

G++ 12.3.1.

### Bug prevalence

100% when using G++ 8.2.0

### GitHub hash of the SDK that was being used

a068855d27b42b2c17d38b3a8a9d30d672163aee

### Platform

other

### Platform Version(s)

Linux

### Anything else?

I suspect this is due to some C++ STL functionality in 8.2.0 used to populate `const chip::Access::CheckData [57]{ ... }` is not correctly `costexpr` valid/safe.

gerickson, 2024-09-25:
@kiel-apple, any thoughts on this one?

andy31415, 2024-10-24:
https://gcc.gnu.org/projects/cxx-status.html#cxx17 seems to say ` Some C++17 features are available since GCC 5, but support was experimental and the ABI of C++17 features was not stable until GCC 9` so I wonder if gcc 8.2 is supporting C++17 (which is the minimum requirement of the SDK)