project-chip/connectedhomeip #35747: [CERT-TEST-FAILURE] [TC-TSTAT-1.1] ClusterRevision gets checked against the wrong value

bauerschwan, 2024-09-24:
### Feature Area

Other

### Test Case

TC-TSTAT-1.1

### Reproduction steps

TC-TSTAT-1.1 step 2 checks against a wrong value for the ClusterRevision attribute.
Expected value from spec: 8
Actual value used in the test: 7

### Bug prevalence

with every run of that test case

### GitHub hash of the SDK that was being used

17b1a38e909e7874593bcb87c31be03a5866f1d4

### Platform

nrf

ReneJosefsen, 2024-09-24:
To make sure the information is captured here as well, this test is from my understanding covered by the latest IDM tests, so I suggest the fix here is to have the test case removed from the test plan and have the YAML removed from this repo.

raul-marquez-csa, 2024-10-22:
> To make sure the information is captured here as well, this test is from my understanding covered by the latest IDM tests, so I suggest the fix here is to have the test case removed from the test plan and have the YAML removed from this repo.

Test case will be removed by this PR
[Removes test case scripts #36185
](https://github.com/project-chip/connectedhomeip/pull/36185)

closed 2024-10-23 by raul-marquez-csa