project-chip/connectedhomeip #35748: [CERT-TEST-FAILURE] [TC-TSTAT-2.2] several test steps use hard-coded values for writing instead of the values read from another attribure

bauerschwan, 2024-09-24:
### Feature Area

Other

### Test Case

TC-TSTAT-1.1

### Reproduction steps

Perform test case with a DUT which uses different `AbsMinHeatSetpointLimit` values than the values hard-coded in the YAML script (f.e. 1000 for `AbsMinHeatSetpointLimit`)

Expected result:
Step 6a should use values derived from `AbsMinHeatSetpointLimit` and should succeed

Actual result:
Step 6a uses a hard-coded value (800) which is not in range of [`AbsMinHeatSetpointLimit`... `AbsMaxHeatSetpointLimit`]

### Bug prevalence

every time this test case is performed

### GitHub hash of the SDK that was being used

17b1a38e909e7874593bcb87c31be03a5866f1d4

### Platform

nrf

ReneJosefsen, 2024-10-03:
As a nitpick here, the test case number is wrong in the description.

raul-marquez-csa, 2024-10-07:
@bauerschwan, please provide the following:

- Full list of steps with the stated issue
- App used (DUT)
- Command used to run the test
- Default PICS or custom PICS?
- Logs describing the failure/error

raul-marquez-csa, 2024-10-07:
I have run the test several times without failure, cannot reproduce the bug.
Without the info from my previous comment, I ran it like this:
`./scripts/tests/run_test_suite.py --log-level debug --chip-tool out/linux-x64-chip-tool/chip-tool --target Test_TC_TSTAT_2_2 run --all-clusters-app out/linux-x64-thermostat/thermostat-app`

However, some observations:
> Step 6a uses a hard-coded value (800) which is not in range of [AbsMinHeatSetpointLimit... AbsMaxHeatSetpointLimit]

In the test plan and the YAML, the max value used is **MaxHeatSetpointLimit**, not **AbsMaxHeatSetpointLimit**

**Test Plan**
https://github.com/CHIP-Specifications/chip-test-plans/blob/b9cb9be6fa63c16433928b0ffb4d53d2ee04343c/src/cluster/thermostat.adoc#L759-L762
[image]

**YAML**
https://github.com/project-chip/connectedhomeip/blob/52a5589b11f534e8e6c75465a4c5942ec4500a36/src/app/tests/suites/certification/Test_TC_TSTAT_2_2.yaml#L842-L884

Also, as stated by Rene, the test case name in the description doesn't match the one in the title, are the details of the issue correct in the description?

Thanks.

bauerschwan, 2024-10-08:
I changed the wrong test case name. Sorry for the confusion.

fessehaeve, 2024-10-08:
@raul-marquez-csa the issue is not with the first 6a step - but the 3rd and 4th steps labeled 6a.
The 3rd "step 6a" attempts to write 800 to MinHeatSetpointLimit regardless of the AbsMinHeatSetpointLimit value. 

**YAML**
https://github.com/project-chip/connectedhomeip/blob/52a5589b11f534e8e6c75465a4c5942ec4500a36/src/app/tests/suites/certification/Test_TC_TSTAT_2_2.yaml#L867-L885

You are not running to this issue since you are running the test against all-clusters-app which has AbsMinHeatSetpointLimit value of 700.

CKnoerzer, 2024-10-08:
@raul-marquez-csa Is it possible to have in line 875 and line 884 instead of:
`           value: 800 `
something like:
`           value: AbsMinHeatSetpointLimitValue+100 `

raul-marquez-csa, 2024-10-21:
A python implementation of this test case has been developed in this PR
[[HVAC] Implement TC-TSTAT-2.2 #36023
](https://github.com/project-chip/connectedhomeip/pull/36023)

@bauerschwan, is this update still a priority?

bauerschwan, 2024-10-23:
As soon as https://github.com/project-chip/connectedhomeip/pull/36023 is merged, this issue can be closed

PR #36023 (2025-02-11): [HVAC] Implement TC-TSTAT-2.2
This provides a Python implementation of TC-TSTAT-2.2 to replace the unused TC-TSTAT-2.2 YAML test.

Fixes #35748

#### Testing

This should restore the setpoint values after the test, allowing for multiple runs.
closed 2025-02-11 by mergify[bot]