project-chip/connectedhomeip #35766: [Workflow] 'Run framework tests (leaks, ENABLE_LEAK_DETECTION=1)' failure

lboue, 2024-09-25:
### Workflow issues

Hello,

Job [30627148241](https://github.com/project-chip/connectedhomeip/actions/runs/11027733002/job/30627148241?pr=35392) for 
PR #35392](https://github.com/project-chip/connectedhomeip/pull/35392) is failing after 24m

Could you fix this CI pipeline?

Regards

### Platform

all

bzbarsky-apple, 2024-09-25:
@lboue I'm looking into that, but the job there is picking up a real (albeit intermittent?) leak....

lboue, 2024-09-25:
> @lboue I'm looking into that, but the job there is picking up a real (albeit intermittent?) leak....

Why does it work for all the other pipelines and not this one?

bzbarsky-apple, 2024-09-25:
This is the one pipeline that does a leak check.

bzbarsky-apple, 2024-09-26:
For now, fixed by https://github.com/project-chip/connectedhomeip/pull/35782

closed 2024-09-26 by bzbarsky-apple