project-chip/connectedhomeip #35796: [Feature]thermostat siri automatic problem！big problem ⚠️

Ko-Hon, 2024-09-26:
### Feature description

I don't understand why you want to make a logic of controlling the temperature internally? Many times it is not needed, and developers can do this simple logic themselves.
For example, if the current temperature is 25 degrees Celsius, I will use Siri and tell him to adjust it to 26 degrees Celsius, and he will set the thermostat to 26 degrees Celsius for heating, and oh my God, 26 degrees Celsius for cooling and heating is still very different, and I hope that this option will be available in the future.

We also have certified versions of our products, and these kinds of issues really affect the sales, and your reputation.

I hope you guys will reconsider and get back to me，thanks

### Platform

esp32

### Platform Version(s)

all

bzbarsky-apple, 2024-09-26:
@Ko-Hon Please clearly describe:

1. The expected behavior.
2. The observed behavior and how it differs from the expected one.

Because you say "I hope that this option will be available in the future" and I have no idea what "option" you are talking about, or what the problem you are facing is.

Ko-Hon, 2024-09-26:
When using the voice assistant Siri, when the current temperature is lower than the set temperature (e.g., if the current ambient temperature is 25 degrees Celsius, and the current set temperature is 23 degrees Celsius, you need to set it to 26 degrees Celsius (Siri Voice) The accessory will automatically adjust the heating mode to 26 degrees Celsius)

I'd like this feature to give engineers a choice, rather than forcing the logic to be kept in it.

If needed, I can record a video for you.

thanks for your reply

bzbarsky-apple, 2024-09-28:
That comment is self-contradictory: "current temperature is lower than the set temperature" and "current ambient temperature is 25 degrees Celsius, and the current set temperature is 23 degrees Celsius".

What is the issue _in the Matter SDK_ involved here?

andy31415, 2024-10-24:
@Ko-Hon please describe what the desired behavior of the `Matter SDK` is here? What is the existing behavior and what is the desired behavior?