project-chip/connectedhomeip #35798: [Feature] I need to build an app to control matter devices by Unity Engine

ehabtawfikelbwab, 2024-09-26:
### Feature description

Hello,

I have developed an application using the Unity engine that manages MQTT, WebSockets, and HTTP requests (both TCP and UDP). However, I have been unable to find an SDK or relevant documentation for integrating Matter with Unity.

Could you assist me on how to proceed with Matter integration in Unity?

Your help would be greatly appreciated.

### Platform

other

### Platform Version(s)

0

### Anything else?

Hello,

I have developed an application using the Unity engine that manages MQTT, WebSockets, and HTTP requests (both TCP and UDP). However, I have been unable to find an SDK or relevant documentation for integrating Matter with Unity.

Could you assist me on how to proceed with Matter integration in Unity?

Your help would be greatly appreciated.

Jardaliao, 2024-09-30:
You may want to develop your  application as a Matter Hub device, which is able to pair and control matter devices, and subscribe attributes of matter devices as well. So that your application can convert the interaction with matter devices into MQTT, WebSocket, etc. to interact with the outside world.

As for how to develop your application into a matter hub device, there seems to be no out-of-the-box documentation.
My experience is to refer to the examples under the example directory, especially chip-tool. Chip-tool is mainly used as a client to pair devices and send commands, much like a matter hub.

ehabtawfikelbwab, 2024-09-30:
I need to develop an application similar to Google Home that controls Matter devices directly, without requiring a hub. 
My Matter devices are based on ESP32, and I want to control them directly without a hub. 

After searching extensively, I haven't found any existing solutions for integrating Matter with Unity. 

Unity works well with protocols like MQTT, WebSockets, and HTTP requests (both TCP and UDP).

Unity’s big platform and large community make adding Matter support a valuable and impactful solution.

ehabtawfikelbwab, 2025-04-28:
Any updates ?

We strongly believe that Unity's multi-platform power, combined with Matter's universal smart home protocol.

Therefore, we kindly request:

An official Unity Matter Plugin or C# SDK that abstracts device discovery, pairing, and communication.

Native support for Matter device interactions (controllers, bridges, endpoints) inside Unity scripting.

Examples or templates for Matter-integrated Unity projects.

Many businesses and developers — including myself — would immediately adopt such support in building smarter, interoperable, and future-proof applications.