project-chip/connectedhomeip #35804: [Build][Silabs] ../../../examples/lit-icd-app/silabs/src/AppTask.cpp:107:33: error: 'class chip::Server' has no member named 'GetICDManager'

lboue, 2024-09-26:
### Build issue(s)

Hi,

I am facing an issue building [Matter EFR32 Lit ICD Example](https://github.com/project-chip/connectedhomeip/tree/master/examples/lit-icd-app/silabs) example.

1. cd ~/connectedhomeip
2. ./scripts/examples/gn_silabs_example.sh ./examples/lit-icd-app/silabs/ ./out/lit-icd-app BRD4187C

```
-I../../../examples/lit-icd-app/silabs/third_party/connectedhomeip/examples/common/QRCode/repo/c -I../../../examples/lit-icd-app/silabs/third_party/connectedhomeip/examples/providers -c ../../../examples/lit-icd-app/silabs/src/AppTask.cpp -o obj/src/matter-silabs-lit-icd-example.out.AppTask.cpp.o
../../../examples/lit-icd-app/silabs/src/AppTask.cpp: In static member function 'static void AppTask::AppTaskMain(void*)':
../../../examples/lit-icd-app/silabs/src/AppTask.cpp:107:33: error: 'class chip::Server' has no member named 'GetICDManager'
  107 |     chip::Server::GetInstance().GetICDManager().RegisterObserver(&sAppTask);
      |                                 ^~~~~~~~~~~~~
At global scope:
cc1plus: note: unrecognized command-line option '-Wno-unknown-warning-option' may have been intended to silence earlier diagnostics
ninja: build stopped: subcommand failed.
```

### Platform

efr32

### Anything else?

[lit-icd-app_build.log](https://github.com/user-attachments/files/17155489/lit-icd-app_build.log)

bzbarsky-apple, 2024-09-26:
That sounds like `CHIP_CONFIG_ENABLE_ICD_SERVER` is not set true in your case.  Why is it not set true?  Are you missing a `chip_enable_icd_server = true` in the relevant `args.gni` file?

lboue, 2024-09-26:
> That sounds like `CHIP_CONFIG_ENABLE_ICD_SERVER` is not set true in your case. Why is it not set true? Are you missing a `chip_enable_icd_server = true` in the relevant `args.gni` file?

It should be enabled here:

https://github.com/project-chip/connectedhomeip/blob/7eb96cd1e862707c335f09cff826e52ec2134baf/examples/lit-icd-app/silabs/build_for_wifi_args.gni#L29

bzbarsky-apple, 2024-09-26:
Well, are those ending up being loaded by something reachable from `src/app/icd/server/BUILD.gn`?

closed 2024-12-13 by lboue