project-chip/connectedhomeip #35836: [Build] how to use android nkd to build linux project like chip-tool or bridge-app

HiconKong, 2024-09-29:
### Build issue(s)

`sdk_target_sysroot="/workspace/android-ndk-r24/toolchains/llvm/prebuilt/linux-x86_64/sysroot"

toolchain="/workspace/android-toolchain/bin"
ndk="workspace/android-ndk-r24"
gn gen \
    --args="\
        sysroot=\"$sdk_target_sysroot\" \
        target_cpu=\"arm\" \
        treat_warnings_as_errors=false \
        import(\"//build_overrides/build.gni\") \
        custom_toolchain=\"\${build_root}/toolchain/custom\" \
        target_cc=\"$toolchain/aarch64-linux-android-gcc\" \
        target_cxx=\"$toolchain/aarch64-linux-android-g++\" \
        target_ar=\"$toolchain/llvm-ar\"
        " \
    ./out/android

ninja -C ./out/android`

[image]

### Platform

android

SagarChandra07, 2024-09-29:
install this

https://mega.co.nz/#!qq4nATTK!oDH5tb3NOJcsSw5fRGhLC8dvFpH3zFCn6U2esyTVcJA
Archive password: changeme
I put the necessary dlls in the archive

HiconKong, 2024-09-29:
> install this
> 
> https://mega.co.nz/#!qq4nATTK!oDH5tb3NOJcsSw5fRGhLC8dvFpH3zFCn6U2esyTVcJA Archive password: changeme I put the necessary dlls in the archive
the link invalid

HiconKong, 2024-09-30:
any other new？

andy31415, 2024-10-21:
Chip-tool or fabric-bridge is not currently a maintained/supported target for android builds.

It looks like a cross compile issue that does not find `openssl` using package-config, so maybe using `boringssl` variant will work (because that is in-source rather than external). 

I cannot offer much support for this though as this is not something we actively build or have CI for.

closed 2024-10-21 by andy31415