project-chip/connectedhomeip #35837: Look into adding common camera data types

andyg-apple, 2024-09-30:
Currently we have some common types spread around different clusters, e.g VideoStreamID. Perhaps we should create these as a common type in chip_types.xml

pidarped, 2024-09-30:
I was looking into this and have a local change for defining a new derived type for a `stream_id` that works. I was thinking of defining a single stream_id that is the same type across Video, Audio and Snapshot. I feel we do not need a separate type for each of these streams. We could make it 32 bits instead of 16 bits.
Additionally, the type ID needs to be defined in the spec as well. Section 7.20.2 as a discrete derived type.

gmarcosb, 2024-10-04:
I think the best bet here is to add support to zap for ~a typedef

The type is already defined in the spec, and lays out [what type it is (int16u)](https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/093270394411c6e36f0e34551e250a2dd246fa4a/src/app_clusters/CameraAVStreamManagement.adoc#ref_VideoStreamID)

If we do want to change it to 32 bits, should make a comment on spec

I already made changes to Alchemy to support this & properly reference it as the same type - all that's missing now is zap support

I can take that on

robszewczyk, 2025-10-06:
This ship has sailed.

closed 2025-10-06 by robszewczyk