project-chip/connectedhomeip #35870: TypeError: Couldn't build proto file into descriptor pool: duplicate symbol 'pw.rpc.EchoMessage'

VaishaliAvhale, 2024-10-02:
### Reproduction steps

1. connectedhomeip$ source script/bootstrap.sh
2. connectedhomeip$ source script/activate.sh
1.  Enable RPCs in the build using menuconfig and build and flash the lighting-app/esp32
3. build pigweed/rpc_console and execute chip-console --device /dev/ttyUSB0 command 

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

83159c252587e695cf791dada8eb2db55ffc471d 

### Platform

esp32

### Platform Version(s)

V1.4

### Anything else?

[image]

andy31415, 2024-10-25:
Removed the label - it looks like the `chip-console` has issues here. Unsure about who maintains pw-rpc things though at this point.

andy31415, 2024-10-30:
I was updating some logging and asserts packages, when checking RPC I hit this in #36297

It seems that the issue is that the chip console was not maintained (we seem to have no tests?) and it changged over time:
  - pw_system.console already imports the echo service
  - several arguments removed (no ipython, no output, several flags changed)

I updated it to work in that PR and tested locally on linux so far.