project-chip/connectedhomeip #35898: [Feature] Flush deferred attributes using DeferredAttributePersistenceProvider on Shutdown

lmapii, 2024-10-03:
### Feature description

I'm pretty new to the whole repository so please pardon me if I'm wrong

As far as I can tell, when using the `DeferredAttributePersistenceProvider` via `app::SetAttributePersistenceProvider` any deferred attribute that is pending will not be persisted in case of a shutdown, e.g., a controlled shutdown due to a software update. It is also not easily possible to, e.g., busy-wait on the attribute provider to delay a shutdown until all attributes have been stored.

It would be great if the `Shutdown` method in the `DefaultAttributePersistenceProvider` would consider writing all deferred attributes.

### Platform

all

### Platform Version(s)

N/A

Damian-Nordic, 2024-10-06:
@lmapii 
Hi, what you wrote seems correct. :)

When adding `DeferredAttributePersistenceProvider` I imagined it would be mostly used for attributes like the light bulb brightness, which are:
1. Fast-changing in certain periods of time and storing each increment might impact the storage device lifetime.
2. Non-critical, meaning that it's OK that in rare circumstances an attribute value won't be saved before a reboot.

But I agree there are probably cases where `DeferredAttributePersistenceProvider::Shutdown()` might improve the user experience. Would you be able to contribute such a change?

I'm not sure though if it's easy to plug that method in all _controlled shutdown_ paths as there seems to be no common, platform-independent code for handling reboots in the SDK, but if you can e.g. modify `OTAImageProcessor` used by your application, you could probably flush deferred attribute writes from there.

lmapii, 2024-10-07:
Keeping the implementation as is makes perfect sense to me - I fear that I'm not far enough into the project yet to really make a change, maybe that changes in 2025. By the end of 2025 I hope to have a better idea of the flash wear that I'll be looking at, but I have some more pressing issues still :D

Coming from a different perspective I had the idea to, e.g., store fast changing attributes only every minute or ten: Not storing the correct lightning level in case of a power outage wouldn't be a problem, but if there was a software update it'd make sense not to lose any settings.

This could, of course, also be implemented using an "External"ly stored attribute in case I find out how to hook into the shutdown procedure.

Should we keep the issue open? I could also close it and open a new one with a possible implementation once/if I get the chance to have a better look.

bzbarsky-apple, 2024-10-07:
Realistically, if whatever OTA shutdown does calls `Server::Shutdown` then it will reach the attribute persistence provider's Shutdown.

If OTA shutdown does not call `Server::Shutdown`, then perhaps it should?

lmapii, 2024-10-07:
As far as I can tell the call sequence does reach `Server::Shutdown` and therefore also calls the persistence provider's (empty) `Shutdown` method, yes.