project-chip/connectedhomeip #35914: [Build] Failed to build linux all-cluster-app.

jadhavrohit924, 2024-10-04:
### Build issue(s)

- I have activated the CHIP environment and ran `./scripts/examples/gn_build_example.sh examples/all-clusters-app/linux out/host` to build the linux all-clusters-app.

I got the following error
> connectedhomeip/out/host/../../examples/all-clusters-app/linux/third_party/connectedhomeip/third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py", line 202, in <module>
    module: str | None,
TypeError: unsupported operand type(s) for |: 'type' and 'NoneType'

I found that `str | None` is introduced in Python 3.10 and I'm using Intel Mac which forces me to use Python 3.9.5+chromium.19.

Is there any solution to this problem?

cc: @dhrishi

andy31415, 2024-10-07:
@jadhavrohit924 we require python 3.10 or above currently.
If you are on darwin, python 3.11 should be auto-pulled. If you are on linux, we use system python and that has to be 3.10 or above.

andy31415, 2024-10-07:
ah, I see `using Intel Mac`

bootstrap should have pulled a python 3.11 for you

jadhavrohit924, 2024-10-08:
Thanks. It works now.

closed 2024-10-08 by jadhavrohit924