project-chip/connectedhomeip #35934: [BUG] android - handleNotifyUpdateApplied is never called

mzakharocsc, 2024-10-07:
### Reproduction steps

Compile and run android chip-tool on master branch bcc1f1b3b8243326ec1818b0d1b84c37464bd0f6
Perform OTA update from OTA requester node fw version 1 to 2. 
Node in this case is esp32 light example matter project.
Expect [handleNotifyUpdateApplied](https://github.com/project-chip/connectedhomeip/blob/bcc1f1b3b8243326ec1818b0d1b84c37464bd0f6/examples/android/CHIPTool/app/src/main/java/com/google/chip/chiptool/clusterclient/OtaProviderClientFragment.kt#L621C18-L621C43) to execute once the node reboots to new firmware. This callback never fires.

May be of relevance, once node boots from OTA image update, the following logs are printed to the console:
```
E (47726) chip[DIS]: OperationalSessionSetup[1:000000000001B669]: operational discovery failed: 32
E (47736) chip[SWU]: Failed to connect to node 0x000000000001B669: 32
```
1B669 in this case is the node ID of the controller OTA Provider.

Hoever, with example linux OTA-Provider,  NotifyUpdateApplied works as expected.

### Bug prevalence

this happens every time OTA is performed

### GitHub hash of the SDK that was being used

bcc1f1b3b8243326ec1818b0d1b84c37464bd0f6

### Platform

android

Trishia, 2024-10-16:
@mzakharocsc 
Did you try this below ? 
`./chip-ota-requestor-app --KVS /tmp/chip-ota-requestor  --autoApplyImage` 

In case of using above '--autoApplyImage`, I have got NotifyUpdateApplied on android controller. 
See android logs: 
```
12:39:26.993  2141-2297  DMG                      D  InvokeRequestMessage =
12:39:26.993  2141-2297  DMG                      D  {
12:39:26.993  2141-2297  DMG                      D  	suppressResponse = false, 
12:39:26.993  2141-2297  DMG                      D  	timedRequest = false, 
12:39:26.993  2141-2297  DMG                      D  	InvokeRequests =
12:39:26.993  2141-2297  DMG                      D  	[
12:39:26.993  2141-2297  DMG                      D  		CommandDataIB =
12:39:26.993  2141-2297  DMG                      D  		{
12:39:26.993  2141-2297  DMG                      D  			CommandPathIB =
12:39:26.993  2141-2297  DMG                      D  			{
12:39:26.993  2141-2297  DMG                      D  				EndpointId = 0x0,
12:39:26.993  2141-2297  DMG                      D  				ClusterId = 0x29,
12:39:26.993  2141-2297  DMG                      D  				CommandId = 0x4,
12:39:26.993  2141-2297  DMG                      D  			},
12:39:26.993  2141-2297  DMG                      D  			
12:39:26.993  2141-2297  DMG                      D  			CommandFields = 
12:39:26.993  2141-2297  DMG                      D  			{
12:39:26.993  2141-2297  DMG                      D  				0x0 = [
12:39:26.993  2141-2297  DMG                      D  						0x9d, 0x57, 0x01, 0x4d, 0x46, 0x5f, 0x33, 0xc8, 0xb1, 0xf4, 0xc2, 0x26, 0x95, 0x9a, 0x13, 0x31, 0x80, 0x66, 0xad, 0x18, 0x85, 0x27, 0xee, 0x5e, 0x38, 0xae, 0x8a, 0x44, 0x7a, 0x40, 0xe3, 0xa1, 
12:39:26.993  2141-2297  DMG                      D  				] (32 bytes)
12:39:26.993  2141-2297  DMG                      D  				0x1 = 2 (unsigned), 
12:39:26.993  2141-2297  DMG                      D  			},
12:39:26.993  2141-2297  DMG                      D  		},
12:39:26.994  2141-2297  DMG                      D  		
12:39:26.994  2141-2297  DMG                      D  	],
12:39:26.994  2141-2297  DMG                      D  	
12:39:26.994  2141-2297  DMG                      D  	InteractionModelRevision = 11
12:39:26.994  2141-2297  DMG                      D  },
12:39:26.994  2141-2297  DMG                      D  Received command for Endpoint=0 Cluster=0x0000_0029 Command=0x0000_0004
12:39:26.994  2141-2297  ZCL                      I  OTA Provider received NotifyUpdateApplied
12:39:26.994  2141-2297  ZCL                      D    Update Token: 32
12:39:26.994  2141-2297  ZCL                      D    Software Version: 2
```

joonhaengHeo, 2024-10-24:
@mzakharocsc 
When this issue occurs, can you upload Android Full Log? We also need to check if mDNS of Android Framework works properly.

mzakharocsc, 2024-10-24:
Attached node and logcat logs. This occurs every time update is upplied to the node.
[logcat.txt](https://github.com/user-attachments/files/17508589/logcat.txt)
[dut.log](https://github.com/user-attachments/files/17508591/dut.log)

mzakharocsc, 2024-10-24:
> Did you try this below ?
> `./chip-ota-requestor-app --KVS /tmp/chip-ota-requestor --autoApplyImage`

I am struggling getting `chip-ota-rquestor-app` provisioned. Android CHIP tool option `PROVISION CHIP DEVICE WITH CUSTOM FLOW` does not seem to be doing anything after I scan the QR code. Is there a trick to it? For me, only `PROVISION CHIP DEVICE WITH WIFI` works for my esp32 node.

joonhaengHeo, 2025-04-02:
I think that the failure occurs because Operational Discovery is disabled during OTA operation. Can you tell me why Operational Discovery is performed?