project-chip/connectedhomeip #35972: [BUG] Example tv-app fails TC_DeviceBasicComposition test

soares-sergio, 2024-10-08:
### Reproduction steps

How to reproduce:
1. Run the `TC_BasicDeviceComposition.py` against the tv-app. Command:
```
scripts/run_in_python_env.sh out/python_env './scripts/tests/run_python_test.py --app ./out/linux-x64-tv-app-no-ble-clang-boringssl/chip-tv-app --app-args "--trace-to json:log" --script src/python_testing/TC_DeviceBasicComposition.py --script-args "--qr-code MT:-24J0AFN00KA0648G00"'
```

Full list of failures: 
```
chip-tv-app,"  Problem: ProblemSeverity.ERROR
  test_name: test_TC_IDM_10_1
  location:
  Endpoint: 4,
  Cluster:  29 (0x1d) Descriptor
  Attribute:65532 (0xfffc)
  problem: Did not find mandatory global Attribute FeatureMap 65532, 0xFFFC on Endpoint 4, Cluster Descriptor (29, 0x001D)"
chip-tv-app,"  Problem: ProblemSeverity.ERROR
  test_name: test_TC_IDM_10_1
  location:
  Endpoint: 4,
  Cluster:  1293 (0x50d) ApplicationBasic
  Attribute:65532 (0xfffc)
  problem: Did not find mandatory global Attribute FeatureMap 65532, 0xFFFC on Endpoint 4, Cluster ApplicationBasic (1293, 0x050D)"
chip-tv-app,"  Problem: ProblemSeverity.ERROR
  test_name: test_TC_IDM_10_1
  location:
  Endpoint: 4,
  Cluster:  1292 (0x50c) ApplicationLauncher
  Attribute:65532 (0xfffc)
  problem: Did not find mandatory global Attribute FeatureMap 65532, 0xFFFC on Endpoint 4, Cluster ApplicationLauncher (1292, 0x050C)"
chip-tv-app,"  Problem: ProblemSeverity.ERROR
  test_name: test_TC_IDM_10_1
  location:
  Endpoint: 4,
  Cluster:  1294 (0x50e) AccountLogin
  Attribute:65532 (0xfffc)
  problem: Did not find mandatory global Attribute FeatureMap 65532, 0xFFFC on Endpoint 4, Cluster AccountLogin (1294, 0x050E)"
chip-tv-app,"  Problem: ProblemSeverity.ERROR
  test_name: test_TC_IDM_10_1
  location:
  Endpoint: 4,
  Cluster:  1285 (0x505) TargetNavigator
  Attribute:65532 (0xfffc)
  problem: Did not find mandatory global Attribute FeatureMap 65532, 0xFFFC on Endpoint 4, Cluster TargetNavigator (1285, 0x0505)"
chip-tv-app,"  Problem: ProblemSeverity.ERROR
  test_name: test_TC_IDM_10_1
  location:
  Endpoint: 5,
  Cluster:  29 (0x1d) Descriptor
  Attribute:65532 (0xfffc)
  problem: Did not find mandatory global Attribute FeatureMap 65532, 0xFFFC on Endpoint 5, Cluster Descriptor (29, 0x001D)"
chip-tv-app,"  Problem: ProblemSeverity.ERROR
  test_name: test_TC_IDM_10_1
  location:
  Endpoint: 5,
  Cluster:  1293 (0x50d) ApplicationBasic
  Attribute:65532 (0xfffc)
  problem: Did not find mandatory global Attribute FeatureMap 65532, 0xFFFC on Endpoint 5, Cluster ApplicationBasic (1293, 0x050D)"
chip-tv-app,"  Problem: ProblemSeverity.ERROR
  test_name: test_TC_IDM_10_1
  location:
  Endpoint: 5,
  Cluster:  1292 (0x50c) ApplicationLauncher
  Attribute:65532 (0xfffc)
  problem: Did not find mandatory global Attribute FeatureMap 65532, 0xFFFC on Endpoint 5, Cluster ApplicationLauncher (1292, 0x050C)"
chip-tv-app,"  Problem: ProblemSeverity.ERROR
  test_name: test_TC_IDM_10_1
  location:
  Endpoint: 5,
[... 154 lines cut ...]
  Cluster:  1289 (0x509) KeypadInput
  Attribute:65533 (0xfffd)
  problem: Failed validation of value on Endpoint 7, Cluster KeypadInput (1289, 0x0509), Attribute ClusterRevision 65533, 0xFFFD: Value ValueDecodeFailure(TLVValue=None, Reason=InteractionModelError(<Status.Failure: 1>)) is not an integer or uint (decoded type: <class 'chip.clusters.Attribute.ValueDecodeFailure'>)"
chip-tv-app,"  Problem: ProblemSeverity.ERROR
  test_name: test_TC_IDM_10_1
  location:
  Endpoint: 7,
  Cluster:  1292 (0x50c) ApplicationLauncher
  Attribute:65533 (0xfffd)
  problem: Failed validation of value on Endpoint 7, Cluster ApplicationLauncher (1292, 0x050C), Attribute ClusterRevision 65533, 0xFFFD: Value ValueDecodeFailure(TLVValue=None, Reason=InteractionModelError(<Status.Failure: 1>)) is not an integer or uint (decoded type: <class 'chip.clusters.Attribute.ValueDecodeFailure'>)"
```

Looking at the ZAP file, I can see some of the feature map there (https://github.com/project-chip/connectedhomeip/blob/master/examples/tv-app/tv-common/tv-app.zap#L7046). But it looks like this app has dynamic endpoints and the feature map is not added there, for example:  https://github.com/project-chip/connectedhomeip/blob/master/examples/tv-app/tv-common/src/AppTv.cpp#L338

### Bug prevalence

always

### GitHub hash of the SDK that was being used

latest

### Platform

other, core

chrisdecenzo, 2024-10-08:
Looks like the feature map attribute (0xfffc) is missing on a handful of the clusters of the example tv-app

lazarkov, 2024-10-17:
@soares-sergio 

Do these tests fail when you read each of the values for a given endpoint?
How does the code know endpoint 4,5,6,7 is available?

soares-sergio, 2024-10-17:
@lazarkov 

The code for the test is here: https://github.com/project-chip/connectedhomeip/blob/master/src/python_testing/TC_DeviceBasicComposition.py#L292

It basically performs a wildcard read of all attributes in all endpoints and checks that everything required in the spec is present.

I wonder if you're just missing `DECLARE_DYNAMIC_ATTRIBUTE(Channel::Attributes::FeatureMap::Id, BITMAP32, 4, 0),                       /* FeatureMap */` in a few of the clusters in https://github.com/project-chip/connectedhomeip/blob/master/examples/tv-app/tv-common/src/AppTv.cpp#L338 ?

lazarkov, 2024-10-17:
@soares-sergio thanks for the info! That might be the case. I've pinged [TCL]Zeng Li, who showed interest in resolving this issue.

lazarkov, 2024-10-17:
note that similar issue has been resolved with this PR (https://github.com/project-chip/connectedhomeip/pull/31863)