project-chip/connectedhomeip #35983: [BUG] asyncio timeout error experienced when attempting to get window_status from endpoint after commissioning window expires

j-ororke, 2024-10-08:
### Reproduction steps

**Test Steps:**
1. TH_CR1 starts a commissioning process with DUT_CE
2. TH_CR1 reads the BasicCommissioningInfo attribute from the General Commissioning cluster and saves the MaxCumulativeFailsafeSeconds field as `max_window_duration`.
3a. TH_CR1 opens a commissioning window on DUT_CE using a commissioning timeout of `max_window_duration` using ECM
3b. DNS-SD records shows DUT_CE advertising
3c. TH_CR1 writes and reads the Basic Information Cluster's NodeLabel mandatory attribute of DUT_CE
4. TH creates a controller (TH_CR2) on a new fabric and commissions DUT_CE using that controller. TH_CR2 should commission the device using a different NodeID than TH_CR1.
5. TH_CR1 reads the Fabrics attribute from the Node Operational Credentials cluster using a fabric-filtered read 
6. TH_CR2 reads the Fabrics attribute from the Node Operational Credentials cluster using a fabric-filtered read 
7. TH_CR1 writes and reads the Basic Information Cluster's NodeLabel mandatory attribute of DUT_CE
8. TH_CR2 reads, writes and then reads the Basic Information Cluster's NodeLabel mandatory attribute of DUT_CE
9. TH_CR2 opens a commissioning window on DUT_CE for 180 seconds using ECM
10. Wait for the commissioning window in step 9 to timeout
11. TH_CR2 reads the window status to verify the DUT_CE window is closed

### Bug prevalence

Consistent

### GitHub hash of the SDK that was being used

17b1a38e909e7874593bcb87c31be03a5866f1d4

### Platform

python, raspi, vscode

### Anything else?

**Pre-Requisites:**
> TH_CR1 = Test harness as Commissioner 1
> TH_CR2 = Test harness as Commissioner 2
> DUT = Commissionee

**Expected Result:**
- After test step 9, we wait 180 seconds for ECM to timeout, after timeout period ends during step 11, we should be able to check the window status and receive a kWindowNotOpen Enum in response to inform us that the window is closed

**Actual Result:**
- We receive a timeout response of asyncio.exceptions.TimeoutError (Please see attached log for full failure output).

**Logs Provided from TH:**
- [test_log_debug.txt](https://github.com/user-attachments/files/17300157/test_log_debug.txt)
- [test_log_info.txt](https://github.com/user-attachments/files/17300158/test_log_info.txt)
- [device_output2.txt](https://github.com/user-attachments/files/17300159/device_output2.txt)

**Notes:**
- Issue was noticed while using TH v2.11-beta3.1+fall2024 
- A possible similar issue was reportedly noticed by QA team when running the CADMIN_1_3 yaml script on TH at the following link: [135](https://github.com/project-chip/certification-tool/issues/135)
- We created a try and except workaround currently where we are able to communicate with the endpoint after the initial attempt from test step 11.
- Same issue is noticed on TH and in WSL Linux.

bzbarsky-apple, 2024-10-09:
@cecille This sounds like a test harness issue....

cecille, 2024-10-09:
possibly, possibly not. This came up when automating some tests, it's unclear right now which side is losing the connection, but that seems to be what's happening. I asked Jake to open this to track and share logs.