project-chip/connectedhomeip #35991: [BUG] Unable to get correct MaxIntervalCeiling in OnSubscriptionRequested for ICD

fuxiaoming-lumi, 2024-10-09:
### Reproduction steps

When using ReadHandler::ApplicationCallback to implement a custom strategy, you will not be able to get the MaxIntervalCeiling in OnSubscriptionRequested for ICD devices. 

We only have GetReportingIntervals API and it get mMaxInterval but is overridden

https://github.com/project-chip/connectedhomeip/blob/5224cec13d97a8746a8d1a9da70bc16a859d8ee7/src/app/ReadHandler.cpp#L744 https://github.com/project-chip/connectedhomeip/blob/5224cec13d97a8746a8d1a9da70bc16a859d8ee7/src/app/ReadHandler.cpp#L794

### Bug prevalence

100%

### GitHub hash of the SDK that was being used

5224cec13d97a8746a8d1a9da70bc16a859d8ee7

### Platform

core

bzbarsky-apple, 2024-10-09:
@lpbeliveau-silabs

fuxiaoming-lumi, 2024-10-10:
More details https://csamembers.slack.com/archives/C036A932BQ8/p1728399013259439

closed 2024-10-11 by yunhanw-google

fuxiaoming-lumi, 2024-10-17:
@yunhanw-google @mkardous-silabs @andy31415 I think we also need cherry-pick this commit to v1.4-branch.