project-chip/connectedhomeip #35994: [BUG] chip-tool pairing command does not accept wifi SSID with empty spaces or special char

rquidute, 2024-10-09:
### Reproduction steps

Run chip-tool command and try to paring ble-wifi using a SSID that contains empty spaces or special char
Ex: `./chip-tool pairing ble-wifi 1 "matter chip" chipcert123 20202021 3822` or  `./chip-tool pairing ble-wifi 1 "Haertelgu's Lab WiFi" chipcert123 20202021 3822`

### Related Issue
https://github.com/project-chip/certification-tool/issues/439

### Bug prevalence

always

### GitHub hash of the SDK that was being used

17b1a38e909e7874593bcb87c31be03a5866f1d4

### Platform

python

### Platform Version(s)

N/A

KishokG, 2024-10-09:
@bzbarsky-apple I've tried with single quotation in SSID field, but unfortunately the commissioning was failed. Please correct if i'm wrong. 

**Command used:**
Chip-tool: ./chip-tool pairing ble-wifi 1 "matter'chip" chipcert123 20202021 3822
DUT: ./chip-all-clusters-app --wifi --discriminator 3822

Here is the logs: 
[Ble-wifi_pairing_DUT.txt](https://github.com/user-attachments/files/17312001/Ble-wifi_pairing_DUT.txt)
[Ble-wifi_pairing_TH.txt](https://github.com/user-attachments/files/17312002/Ble-wifi_pairing_TH.txt)

bzbarsky-apple, 2024-10-09:
@KishokG Those logs have nothing to do with chip-tool or the SSID. Your DUT failed to initialize properly:

```
[1728478909.254629][4023:4023] CHIP:IN: Failed to initialize TCP transport: src/inet/TCPEndPointImplSockets.cpp:132: OS Error 0x02000062: Address already in use
[1728478909.254896][4023:4023] CHIP:SVR: ERROR setting up transport: src/inet/TCPEndPointImplSockets.cpp:132: OS Error 0x02000062: Address already in use
```

and then it's not really initialized all the way, so it's not listening for PASE messages, so chip-tool can't establish a PASE session to it.

I suggest checking for other running example apps and making sure to use distinct ports for them or to only be running one at a time.

rquidute, 2024-10-14:
Hi @bzbarsky-apple could you provide an example how can you use `./chip-tool`having a SSID with special char and empty spaces? 
If this command supposed to work properly? 
Chip-tool: `./chip-tool pairing ble-wifi 1 "matter'chip" chipcert123 20202021 3822`

bzbarsky-apple, 2024-10-15:
>  If this command supposed to work properly?

Yes, that should work fine.