project-chip/connectedhomeip #35995: [BUG] Unable to complete the commissioning using CONFIG_MBEDTLS_ATCA_HW_ECDSA_SIGN

sbh82, 2024-10-09:
### Reproduction steps

1. Using Matter SDK 1.3 with ESP-IDF 5.2.1 on ESP32-C6
2. Added **esp-cryptoauthlib** as component, enabled the secure element and enabled `CONFIG_MBEDTLS_ATCA_HW_ECDSA_SIGN` in sdkconfig
3. Used the `light` example as a template to create a simple light on/off switch Matter endpoint
4. Launch the commissioning from a smartphone and a Matter compatible app
5. The commissioning fails reporting this error in ESP log:  `E (117103) chip[CR]: mbedTLS error: BIGNUM - The buffer is too small to write to`

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

82748b9f0e5a4cdc4d10715f10c6160d49d87e75

### Platform

esp32

### Platform Version(s)

ESP-IDF 5.2.1

### Anything else?

Without the step 2, i.e. removing the `esp-cryptoauthlib` and disable the `CONFIG_MBEDTLS_ATCA_HW_ECDSA_SIGN` flag the commissioning works perfectly.
Insepcting the SDK code it seems the error comes from `P256Keypair::ECDSA_sign_msg` method in `config/esp32/third_party/connectedhomeip/src/crypto/CHIPCryptoPALmbedTLS.cpp` file and the call to method `mbedtls_ecdsa_sign` returns the error.
I report the error log:
```
I (117073) chip[ZCL]: OpCreds: Received an AttestationRequest command
I (117083) chip[EM]: <<< [E:27454r S:61872 M:244198318 (Ack:142956033)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [UDP:192.168.1.121:28710] --- Type 0000:10 (SecureChannel:StandaloneAck)
D (117093) chip[EM]: Flushed pending ack for MessageCounter:142956033 on exchange 27454r
E (117103) chip[CR]: mbedTLS error: BIGNUM - The buffer is too small to write to
E (117123) chip[DMG]: Endpoint=0 Cluster=0x0000_003E Command=0x0000_0000 status 0x01 (no additional context)
D (117133) chip[DMG]: Command handler moving to [NewRespons]
```

bzbarsky-apple, 2024-10-09:
I wonder whether this is a similar problem to https://github.com/project-chip/connectedhomeip/issues/35155 ...

@shubhamdp @dhrishi

shubhamdp, 2024-10-10:
@sbh82 will take a look.

giovFr, 2024-12-02:
I disabled CONFIG_ESP_TLS_USE_SECURE_ELEMENT, CONFIG_MBEDTLS_ATCA_HW_ECDSA_SIGN and CONFIG_MBEDTLS_ATCA_HW_ECDSA_VERIFY. The secure element remain active enabling CONFIG_ATCA_MBEDTLS_ECDSA but build isn't successful for idf 5.3.1 and latest esp-cryptoauthlib. 

I found that it is due to the fact that the file "esp-cryptoauthlib/tree/master/cryptoauthlib/lib/mbedtls/atca_mbedtls_wrap.c" is not fully compatible with the mbedTLS library in idf 5.3.1. Resolving the incompatibility (fixing atca_mbedtls_wrap.c) build completes with no errors. Futhermore commissioning works properly.