project-chip/connectedhomeip #36030: [Build] Execution failed for task ':app:processDebugMainManifest'.

gercekcetinkaya, 2024-10-10:
### Build issue(s)

1.
2.
3.
...

### Platform

other

### Anything else?

Hi,

I want to build Chip-tool apk but unfortunately I didnt get it.

2024-10-11 00:59:32 INFO    Building APP android-arm64-chip-tool
2024-10-11 00:59:34 INFO    > Task :app:preBuild UP-TO-DATE
2024-10-11 00:59:34 INFO    > Task :app:preDebugBuild UP-TO-DATE
2024-10-11 00:59:34 INFO    > Task :chip-library:preBuild UP-TO-DATE
2024-10-11 00:59:34 INFO    > Task :chip-library:preDebugBuild UP-TO-DATE
2024-10-11 00:59:34 INFO    > Task :chip-library:compileDebugAidl NO-SOURCE
2024-10-11 00:59:34 INFO    > Task :app:compileDebugAidl NO-SOURCE
2024-10-11 00:59:34 INFO    > Task :chip-library:packageDebugRenderscript NO-SOURCE
2024-10-11 00:59:34 INFO    > Task :app:compileDebugRenderscript NO-SOURCE
2024-10-11 00:59:34 INFO    > Task :app:dataBindingMergeDependencyArtifactsDebug UP-TO-DATE
2024-10-11 00:59:34 INFO    > Task :app:dataBindingMergeGenClassesDebug UP-TO-DATE
2024-10-11 00:59:34 INFO    > Task :app:generateDebugResValues UP-TO-DATE
2024-10-11 00:59:34 INFO    > Task :app:generateDebugResources UP-TO-DATE
2024-10-11 00:59:34 INFO    > Task :chip-library:compileDebugRenderscript NO-SOURCE
2024-10-11 00:59:34 INFO    > Task :chip-library:generateDebugResValues UP-TO-DATE
2024-10-11 00:59:34 INFO    > Task :chip-library:generateDebugResources UP-TO-DATE
2024-10-11 00:59:34 INFO    
2024-10-11 00:59:34 INFO    > Task :chip-library:packageDebugResources
2024-10-11 00:59:34 INFO    Execution optimizations have been disabled for task ':chip-library:packageDebugResources' to ensure correctness due to the following reasons:
2024-10-11 00:59:34 INFO      - Gradle detected a problem with the following location: '/home/gercek/Matter/connectedhomeip/out/android-arm64-chip-tool/generated/res/resValues/debug'. Reason: Task ':chip-library:packageDebugResources' uses this output of task ':app:generateDebugResValues' without declaring an explicit or implicit dependency. This can lead to incorrect results being produced, depending on what order the tasks are executed. Please refer to https://docs.gradle.org/7.1.1/userguide/validation_problems.html#implicit_dependency for more details about this problem.
2024-10-11 00:59:34 INFO      - Gradle detected a problem with the following location: '/home/gercek/Matter/connectedhomeip/out/android-arm64-chip-tool/generated/res/rs/debug'. Reason: Task ':chip-library:packageDebugResources' uses this output of task ':app:compileDebugRenderscript' without declaring an explicit or implicit dependency. This can lead to incorrect results being produced, depending on what order the tasks are executed. Please refer to https://docs.gradle.org/7.1.1/userguide/validation_problems.html#implicit_dependency for more details about this problem.
2024-10-11 00:59:34 INFO    
2024-10-11 00:59:34 INFO    > Task :app:mergeDebugResources UP-TO-DATE
2024-10-11 00:59:34 INFO    > Task :app:dataBindingGenBaseClassesDebug UP-TO-DATE
2024-10-11 00:59:34 INFO    > Task :app:generateDebugBuildConfig
2024-10-11 00:59:34 INFO    > Task :chip-library:writeDebugAarMetadata UP-TO-DATE
2024-10-11 00:59:34 INFO    > Task :app:checkDebugAarMetadata UP-TO-DATE
2024-10-11 00:59:34 INFO    > Task :app:createDebugCompatibleScreenManifests UP-TO-DATE
2024-10-11 00:59:34 INFO    > Task :app:extractDeepLinksDebug UP-TO-DATE
2024-10-11 00:59:34 INFO    > Task :chip-library:extractDeepLinksDebug UP-TO-DATE
2024-10-11 00:59:34 INFO    > Task :chip-library:processDebugManifest
2024-10-11 00:59:34 INFO    > Task :chip-library:compileDebugLibraryResources UP-TO-DATE
2024-10-11 00:59:34 INFO    > Task :chip-library:parseDebugLocalResources UP-TO-DATE
2024-10-11 00:59:34 INFO    > Task :chip-library:generateDebugBuildConfig
2024-10-11 00:59:34 INFO    > Task :chip-library:javaPreCompileDebug UP-TO-DATE
2024-10-11 00:59:34 INFO    > Task :app:javaPreCompileDebug
2024-10-11 00:59:34 INFO    > Task :app:processDebugMainManifest FAILED
2024-10-11 00:59:34 WARNING 
2024-10-11 00:59:34 WARNING FAILURE: Build failed with an exception.
2024-10-11 00:59:34 WARNING 
**2024-10-11 00:59:34 WARNING * What went wrong:
2024-10-11 00:59:34 WARNING Execution failed for task ':app:processDebugMainManifest'.
2024-10-11 00:59:34 WARNING > com.android.manifmerger.ManifestMerger2$MergeFailureException: java.io.FileNotFoundException: /home/gercek/Matter/connectedhomeip/out/android-arm64-chip-tool/intermediates/merged_manifest/debug/AndroidManifest.xml (No such file or directory)**
2024-10-11 00:59:34 WARNING 
2024-10-11 00:59:34 WARNING * Try:
2024-10-11 00:59:34 INFO    
2024-10-11 00:59:34 INFO    Deprecated Gradle features were used in this build, making it incompatible with Gradle 8.0.
2024-10-11 00:59:34 INFO    
2024-10-11 00:59:34 INFO    You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.
2024-10-11 00:59:34 INFO    
2024-10-11 00:59:34 INFO    See https://docs.gradle.org/7.1.1/userguide/command_line_interface.html#sec:command_line_warnings
2024-10-11 00:59:34 INFO    
2024-10-11 00:59:34 INFO    Execution optimizations have been disabled for 1 invalid unit(s) of work during this build to ensure correctness.
2024-10-11 00:59:34 INFO    Please consult deprecation warnings for more details.
2024-10-11 00:59:34 INFO    20 actionable tasks: 6 executed, 14 up-to-date
2024-10-11 00:59:34 WARNING Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
2024-10-11 00:59:34 WARNING 
2024-10-11 00:59:34 WARNING * Get more help at https://help.gradle.org
2024-10-11 00:59:34 WARNING 
2024-10-11 00:59:34 WARNING BUILD FAILED in 2s
Traceback (most recent call last):
  File "/home/gercek/Matter/connectedhomeip/./scripts/build/build_examples.py", line 241, in <module>
    main(auto_envvar_prefix='CHIP')
  File "/home/gercek/Matter/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/gercek/Matter/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/gercek/Matter/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    rv.append(sub_ctx.command.invoke(sub_ctx))
  File "/home/gercek/Matter/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/gercek/Matter/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/gercek/Matter/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/gercek/Matter/connectedhomeip/./scripts/build/build_examples.py", line 231, in cmd_build
    context.obj.Build()
  File "/home/gercek/Matter/connectedhomeip/scripts/build/build/__init__.py", line 73, in Build
    builder.build()
  File "/home/gercek/Matter/connectedhomeip/scripts/build/builders/builder.py", line 107, in build
    self._build()
  File "/home/gercek/Matter/connectedhomeip/scripts/build/builders/android.py", line 482, in _build
    self.gradlewBuildSrcAndroid()
  File "/home/gercek/Matter/connectedhomeip/scripts/build/builders/android.py", line 291, in gradlewBuildSrcAndroid
    self._Execute(
  File "/home/gercek/Matter/connectedhomeip/scripts/build/builders/builder.py", line 112, in _Execute
    self._runner.Run(cmdarray, title=title)
  File "/home/gercek/Matter/connectedhomeip/scripts/build/runner/shell.py", line 73, in Run
    raise Exception('Command %r failed: %d' % (cmd, code))
Exception: Command ['/home/gercek/Matter/connectedhomeip/examples/android/CHIPTool/gradlew', '-p', '/home/gercek/Matter/connectedhomeip/examples/android/CHIPTool', '-PmatterBuildSrcDir=/home/gercek/Matter/connectedhomeip/out/android-arm64-chip-tool', '-PmatterSdkSourceBuild=true', '-PbuildDir=/home/gercek/Matter/connectedhomeip/out/android-arm64-chip-tool', 'assembleDebug'] failed: 1

Why looking for AndroidManifest in debug folder and Why does not exist? 

Is there any one solve this issue?

gercekcetinkaya, 2024-10-11:
A new one ... 

My current kotlin version is **Kotlin version 1.9.0-release-358 (JRE 17.0.12+7-Ubuntu-1ubuntu222.04)**

2024-10-11 02:57:39 WARNING e: file:///home/gercek/Matter/connectedhomeip/examples/android/CHIPTool/app/src/main/java/com/google/chip/chiptool/util/TlvUtil.kt:52:24 Class 'matter.tlv.Tag' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 2.0.0, expected version is 1.8.0.
2024-10-11 02:57:39 WARNING The class is loaded from /home/gercek/.gradle/caches/transforms-3/7ea3a36eaa4bab10a8fcd158662fa9a4/transformed/jetified-libMatterTlv.jar!/matter/tlv/Tag.class
2024-10-11 02:57:39 WARNING 
2024-10-11 02:57:39 WARNING FAILURE: Build failed with an exception.
2024-10-11 02:57:39 WARNING 
2024-10-11 02:57:39 WARNING * **What went wrong:
2024-10-11 02:57:39 WARNING Execution failed for task ':app:compileDebugKotlin'.
2024-10-11 02:57:39 WARNING > A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
2024-10-11 02:57:39 WARNING    > Compilation error. See log for more details**
2024-10-11 02:57:39 WARNING 
2024-10-11 02:57:39 WARNING * Try:
2024-10-11 02:57:39 WARNING Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
2024-10-11 02:57:39 WARNING 
2024-10-11 02:57:39 WARNING * Get more help at https://help.gradle.org
2024-10-11 02:57:39 WARNING 
2024-10-11 02:57:39 WARNING BUILD FAILED in 16s

pachuho, 2024-12-06:
@gercekcetinkaya 

Try changing the Gradle or AGP version set in your Android project. 
It seems that the existing version 7.1.1 is not compatible, causing the issue. 

In my case, I configured it this way and successfully built the project.

[image]

yunhanw-google, 2025-02-03:
we have upgraded the android build with java 17 and gradle 7.3.3, and kotlin 2.0.
https://github.com/project-chip/connectedhomeip/pull/33837, this issue should be resolved, feel free to reopen a new case if this is still problematic. thanks

closed 2025-02-03 by yunhanw-google