project-chip/connectedhomeip #36059: [Device Energy Management Mode Cluster] Device Energy Management Mode Cluster XML has to be updated as per Spec

Ashwinigrl, 2024-10-14:
### Reproduction steps

### Description:

- [ ] 1. Below following `Feature` and `Attributes` are Disallowed for the Device Energy Management Mode Cluster. So it should be removed from the Device Energy Management Mode Cluster device XML.

**Feature:**

1. OnOff  

**Attributes:**

1. StartUpMode
2. OnMode 

**Spec Reference:**
- https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/Mode_DeviceEnergyManagement.adoc#4-features

[image]

[image]

**XML References:**
- https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/device-energy-management-mode-cluster.xml

[image]

[image]

- [ ] 2. In Spec,  Device Energy Management Mode Cluster Revision value has been updated from 1 to 2. So XML has to be updated as per the spec.

**Spec Reference:**
- https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/Mode_DeviceEnergyManagement.adoc#1-revision-history

[image]

**XML References:**
- https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/device-energy-management-mode-cluster.xml#L51

[image]

### Bug prevalence

Whenever we do

### GitHub hash of the SDK that was being used

708558881bf6e5b0dac3c09489124907927b6cd2

### Platform

raspi

### Type

Test Improvement

cecille, 2024-10-16:
Data model XMLs are updated automatically and aligned to spec tags. These data model XMLs will be updated when the spec tag for the board release is tagged.
These data model files are automatically generated and should not be updated by hand. 

They don't need individual issues opened.

Closing this as it will be fixed once the spec is properly tagged and I don't want anyone thinking they need to make hand edits.

bzbarsky-apple, 2024-10-18:
> Data model XMLs are updated automatically and aligned to spec tags

This is not about data model XMLs, which are under `/data_model`.  This is about a file under `/src/app/zap-templates/zcl/data-model/chip`.

jamesharrow, 2024-10-18:
PR here to fix: #36155

PR #36155 (2024-10-21): DEMM, EVSEM, WHM, EEVSE, DEM, EWATERHTR zap XML out of date with 1.4 specs
Fixes #36059 

Regenerated with Alchemy:

`alchemy zap --attribute="in-progress"  --sdkRoot=./connectedhomeip/ --specRoot=./connectedhomeip_spec ./connectedhomeip_spec/src/app_clusters/Mode_EVSE.adoc`

`alchemy zap --attribute="in-progress"  --sdkRoot=./connectedhomeip/ --specRoot=./connectedhomeip_spec/ ./connectedhomeip_spec/src/app_clusters/Mode_WaterHeater.adoc`

`alchemy zap --attribute="in-progress"  --sdkRoot=./connectedhomeip/ --specRoot=./connectedhomeip_spec ./connectedhomeip_spec/src/app_clusters/Mode_DeviceE [... cut]
commit 9ee0499d (2024-10-21): DEMM, EVSEM, WHM, EEVSE, DEM, EWATERHTR zap XML out of date with 1.4 specs (#36155)

* Fixes #36059

* regen_all after change to mode cluster ZCL XML

* Checking on EEVSE the cluster revision was wrong (looks like the rev 4 was backed out after it was felt to still be a draft). Taken in the DEM and WHM latest descriptions based on Alchemy autogen.

* revert description in DEM which was truncated by Alchemy

* regen_all

* Removed src/app/tests/suites/certification/Test_TC_EEVSEM_3_1.yaml, Test_TC_EEVSEM_3_2.yaml, Test_TC_EEVSEM_3_3.yaml which were removed in test plan PR: https://github.com/CH [... cut]
closed 2024-10-21 by andy31415