project-chip/connectedhomeip #36076: FinishCommand

swang4415, 2024-10-15:
### Reproduction steps

I need to send a command to the matter device of the node I selected. First, I set the command path correctly using CommandPathParams, and then called the AddRequestData() function to add the command data. However, when I used the FinishCommand() function, it failed. I don't know how to use this function. Here is my error log:
[image]

### Bug prevalence

Once or twice a week

### GitHub hash of the SDK that was being used

5.0.2

### Platform

esp32

### Type

Common Cluster Logic, Spec Compliance Issue

tehampson, 2024-10-15:
@swang4415, `AddRequestData() calls finish for you. You only need to call Finish if you called Prepare can added the data yourself (which is not recommended)

closed 2024-10-15 by tehampson

swang4415, 2024-10-15:
> @swang4415, `AddRequestData() 会为您调用 finish。如果您调用 Prepare 可以自行添加数据（不推荐），则只需调用 Finish

Do you mean I don't need to use the FinishCommand() function?