project-chip/connectedhomeip #36081: [BUG] Persisted Subscriptions do not cosider new subscriptions with "keepSubscriptions=false"

Apollon77, 2024-10-15:
### Reproduction steps

* Imagine you have a device ( I used the current chip repo "Light" example device on macos) which was successfully operating with a controller and being connected and a subscription is established.
* Now stop the device (device will remember this last subscription)
* wait until the controller is not actively reconnecting to the device anymore (e.g. using latest known address) but wait for the device to be discovered
* now start the device

Now the following happens (depending a bit on the timing):

matter.js as controller (in this case) will see the operational announcement of the device and connect and do a new subscription. According to logs this is finished in

chip log
> [1728996090.398] [93151:8321308] [DMG] Registered a ReadHandler that will schedule a report between system Timestamp: 0x0000000029BBACAD and system Timestamp 0x0000000029BC9325.

matter.js log
> 2024-10-15 14:41:30.398 INFO   InteractionClient    Subscription successfully initialized with ID 1868935302 and maxInterval 60s.

Then at "2024-10-15 14:41:31.411" a new exchange is opened from device side and it starts to push all data anew from this subscription

> 2024-10-15 14:41:31.413 DEBUG  InteractionMessenger Received DataReport chunk with 36 attributes and 0 events, suppressResponse: undefined, moreChunkedMessages: true, subscriptionId: 1868935302
> ...

Additionally also the subscription from before is being tried to push and errors (because we have removed this subscription n matter.js side ... if I would accept it I would have 2 subscriptions!
> 2024-10-15 14:41:31.446 ERROR  ExchangeManager      Invalid Data report with unexpected subscription ID 4118648988

... I can not find this subscription ID in chip log at all :-(

In chip log I also find:
> [1728996090.429] [93151:8321308] [IM] Skip resuming live subscriptionId 1868935302

so he "saw" the new ID but skipped it ...

So it seems that when a new subscription is established with keepSubscriptions=false and somehow the "old" subscription is still there and processed (or scheduled?) that this is not deleted by the new subscription ... and also all attributes are marked as "dirtry" and also pushed again to the already seeded subscription.

Logs:

[matterjs-controller.log](https://github.com/user-attachments/files/17379166/matterjs-controller.log)
[chip.log](https://github.com/user-attachments/files/17379167/chip.log)

### Bug prevalence

always/depends on timing

### GitHub hash of the SDK that was being used

master

### Platform

darwin, other

jtung-apple, 2025-02-28:
@Apollon77 Does this issue still track anything #37741 did not cover?

Apollon77, 2025-03-06:
I need to check that again, lets assume ok for now and I can reopen if needed

closed 2025-03-06 by Apollon77