project-chip/connectedhomeip #36103: [examples/thermostat/linux Commission to CHIP Tool Failed]

skyFzz, 2024-10-16:
### Test issue(s)

When the thermostat is advertising itself, I tried multiple pairing commands but all failed commissioning it to the CHIP Tool server. The thermostat application and the CHIP Tool server are all running on the same Raspberry Pi, connected to Wi-Fi. Commands I tried are `./chip-tool pairing 2 20202021` and `./chip-tool pairing code-wifi 3 wifi pass 34970112332`. By the way I also noted that the `chip_tool_guide.md` is removed from the repo. Is this controller not longer supported?
[image]

Alami-Amine, 2024-10-17:
- chip-tool is still supported, the location of chip_tool_guide.md was moved to: https://github.com/project-chip/connectedhomeip/blob/master/docs/development_controllers/chip-tool/chip_tool_guide.md
- since both your controller and device are in same network (and device), I would use `onnetwork` pairing: 
`./chip-tool pairing onnetwork 7283 20202021`

skyFzz, 2024-10-17:
Yes I have tried `./chip-tool pairing onnetwork id 20202021` multiple times but it all failed. The previous comment shows the error messages.

bzbarsky-apple, 2024-10-18:
The screenshot (next time please attach the logs as text attachments, not as images) shows that the device on the left is already commissioned and operational.  So it can't be commissioned, since it already is.

Also, I suggest not using `pairing onnetwork` which is basically deprecated.  Please use `pairing code`.

closed 2024-10-18 by bzbarsky-apple