project-chip/connectedhomeip #36114: bsp_iot_button_create

swang4415, 2024-10-17:
### Reproduction steps

I now need to define two buttons to send switch and dimming commands respectively, but I encountered some problems when registering the buttons. The log records are shown below:
[image]

My code is this
[image]

Whenever I burn code into esp32c3, it keeps restarting. I don't know what happened. I hope someone can help me solve it. Thanks!!

### Bug prevalence

Once or twice a week

### GitHub hash of the SDK that was being used

5.0.2

### Platform

esp32

### Type

Security Issue

dhrishi, 2024-10-25:
@shubhamdp Please check

shubhamdp, 2024-11-15:
duplicate of #36159