project-chip/connectedhomeip #36134: [BUG] Using commit: 150db874c378d0e167127b1d0c38a3f195c059eb, the repository is clean. Running source ./scripts/activate.sh throws an error.

jczhang777, 2024-10-18:
### Reproduction steps

1.git clone https://github.com/project-chip/connectedhomeip.git
2.git submodule update -f --init --recursive
3.source ./scripts/activate.sh

### Bug prevalence

Whenever  I do this

### GitHub hash of the SDK that was being used

150db874c378d0e167127b1d0c38a3f195c059eb

### Platform

core

### Anything else?

➜  connectedhomeip git:(master) source ./scripts/activate.sh                                                                                                  
2024-10-18 10:32:18,456 Loading extra packages for linux                                                                                                      
2024-10-18 10:32:18,456 Skipping: darwin (i.e. /home/zhang/workplace/matter_all/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/python311.json)
2024-10-18 10:32:18,456 Skipping: windows (i.e. /home/zhang/workplace/matter_all/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/python311.json)
                                                                                                                                                              
  WELCOME TO...                                                                                                                                               
                                                                                                                                                              
         █                                                                                                                                                    
         █                                                                                                                                                    
     ▄   █   ▄                                █     █                                                                                                         
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀                                                                                       
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █                                                                                          
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █                                                                                          
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █                                                                                          
 ▀▀    █   █    ▀▀                                                                                                                                            
                                                                                                                                                              
  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.                                                                                             
                                                                                                                                                              
Downloading and installing packages into local source directory:                                                                                              
                                                                                                                                                              
  Setting up CIPD package manager...done (8.0s)                                                                                                               
  Setting up Project actions........skipped (0.1s)                                                                                                            
  Setting up Python environment.....[\]                                                                                                                       
ninja: Entering directory `/home/zhang/workplace/matter_all/connectedhomeip/.environment/gn_out'                                                              
[1/1220] touch python/obj/third_party/pigweed/repo/pw_metric/metric_service_proto.python._package_metadata._metadata_path_list.txt.stamp                      
[2/1220] touch python/obj/third_party/pigweed/repo/pw_snapshot/snapshot_proto.python._package_metadata._metadata_path_list.txt.stamp                          
[3/1220] touch python/obj/third_party/pigweed/repo/pw_snapshot/py/pw_snapshot_metadata._package_metadata._metadata_path_list.txt.stamp                        
[4/1220] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../third_party/pigweed/repo/pw_cpu_exception_risc_v --default-toolchain=//build/toolchain/linux:linux_x64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --touch python/gen/third_party/pigweed/repo/pw_cpu_exception_risc_v/c
pu_state_protos.python._package_metadata.pw_pystamp --capture-output --python-dep-list-files python/gen/third_party/pigweed/repo/pw_cpu_exception_risc_v/cpu_state_protos.python._package_meta
data_metadata_path_list.txt -- ../../third_party/pigweed/repo/pw_build/py/pw_build/nop.py                                                                     
FAILED: python/gen/third_party/pigweed/repo/pw_cpu_exception_risc_v/cpu_state_protos.python._package_metadata.pw_pystamp                                      
python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../third_party/pigweed/repo/pw_cpu_exception_risc_v --default-toolchain=//buil
d/toolchain/linux:linux_x64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --touch python/gen/third_party/pigweed/repo/pw_cpu_exception_risc_v/cpu_state_
protos.python._package_metadata.pw_pystamp --capture-output --python-dep-list-files python/gen/third_party/pigweed/repo/pw_cpu_exception_risc_v/cpu_state_protos.python._package_metadata_meta
data_path_list.txt -- ../../third_party/pigweed/repo/pw_build/py/pw_build/nop.py                                                                              
Traceback (most recent call last):                                                                                                                            
  File "../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py", line 33, in <module>                                                           
    from pw_build import gn_resolver                                                                                                                          
ModuleNotFoundError: No module named 'pw_build'                                                                                                               
                                                                                                                                                              
During handling of the above exception, another exception occurred:                                                                                           
                                                                                                                                                              
Traceback (most recent call last):                                                                                                                            
  File "../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py", line 38, in <module>                                                           
    import gn_resolver  # type: ignore                                                                                                                        
  File "/home/zhang/workplace/matter_all/connectedhomeip/third_party/pigweed/repo/pw_build/py/pw_build/gn_resolver.py", line 319, in <module>                 
    _Actions = Iterator[tuple[_ArgAction, str]]                                                                                                               
TypeError: 'type' object is not subscriptable                                                                                                                 
[... cut ...]
Traceback (most recent call last):                                                                                                                            
  File "../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py", line 38, in <module>                                                           
    import gn_resolver  # type: ignore                                                                                                                        
  File "/home/zhang/workplace/matter_all/connectedhomeip/third_party/pigweed/repo/pw_build/py/pw_build/gn_resolver.py", line 319, in <module>

jczhang777, 2024-10-18:
Build environment: Ubuntu 20.04, Python version: 3.8.10.

andy31415, 2024-10-18:
@jczhang: Python 3.10 is needed as a minimum to build.
Please update your build environment to meet those requirements.

andy31415, 2024-10-18:
You may also need to clean any existing environment just in case (rm PW_ENVIRONMENT_ROOT and/or .environment) so that bootstrap starts fresh.

jczhang777, 2024-10-21:
@andy31415 Thank you for your reply. I have successfully compiled the code. However, I have some concerns: Ubuntu 20.04's default Python version is 3.8. What is the purpose of the recent upgrade to Matter's environment components that makes it less compatible with Ubuntu 20.04 LTS? Additionally, I'd like to understand what the upgrade plan is for Matter's Python version going forward.

andy31415, 2024-10-21:
We are generally matching https://pigweed.dev/ versioning and they moved to python 3.10. Their intent was to use supported python version however unfortunately they skipped one and we made the team aware of this. In the future the plan would be to only update requirements if python version stops being supported.

For 3.8 in particular, it is already end of life today: https://devguide.python.org/versions/

The only odd bit that we have is that we do not support 3.9 either even though that end of life is 2025.

andy31415, 2024-10-21:
Closing as `requires python 3.10 or above`

closed 2024-10-21 by andy31415