project-chip/connectedhomeip #36141: [Platform] [tv-app][TC-APBSC-9.1] Tested failed

nicelyjust, 2024-10-18:
### Reproduction steps

1. Build the TV app using command:
scripts/build/build_examples.py --target android-arm64-tv-server build
2. Build the tv-casting-app using command:
  $ cd ~/connectedhomeip/examples/tv-casting-app/linux
  $ gn gen out/debug
  $ ninja -C out/debug
3. install and run the platform-app-debug.apk&content-app-debug.apk in the phone
4. Commissioning the tv-casting-app
5. test the case [TC-APBSC-9.1] , occured a failure at step 6 `(UNSUPPORTED_ATTRIBUTE)`

`
> cast cluster applicationbasic read application 0 4
[1729240534.531] [37163:743182] [TOO] Run command failure: examples/chip-tool/commands/common/CHIPCommand.h:253: CHIP Error 0x000000AC: Internal error
Done
> [1729240534.532] [37163:743182] [TOO] nodeId set to 0, using default for fabric 1
[1729240534.532] [37163:743182] [-] Binding type=1 fab=1 nodeId=0xFFFFFFEFFFFFFFFF groupId=65535 local endpoint=0 remote endpoint=1 cluster=0x0000_0000
[1729240534.532] [37163:743182] [-] GetVideoPlayerNodeForFabricIndex nodeId=0xFFFFFFEFFFFFFFFF
[1729240534.532] [37163:743182] [TOO] Sending command to node 0xffffffefffffffff
[1729240534.532] [37163:743182] [CSM] FindOrEstablishSession: PeerId = [1:FFFFFFEFFFFFFFFF]
[1729240534.532] [37163:743182] [CSM] FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1729240534.532] [37163:743182] [DIS] Found an existing secure session to [1:FFFFFFEFFFFFFFFF]!
[1729240534.532] [37163:743182] [DIS] OperationalSessionSetup[1:FFFFFFEFFFFFFFFF]: State change 1 --> 5
[1729240534.532] [37163:743182] [TOO] ModelCommand::OnDeviceConnectedFn
[1729240534.532] [37163:743182] [TOO] Sending ReadAttribute to:
[1729240534.532] [37163:743182] [TOO] 	cluster 0x0000_050D, attribute: 0x0000_0004, endpoint 4
[1729240534.532] [37163:743182] [DMG] SendReadRequest ReadClient[0x138f04080]: Sending Read Request
[1729240534.532] [37163:743182] [DMG] 0 data version filters provided, 0 not relevant, 0 encoded, 0 skipped due to lack of space
[1729240534.533] [37163:743182] [EM] <<< [E:41276i S:17665 M:258475850] (S) Msg TX from C5AA883353531361 to 1:FFFFFFEFFFFFFFFF [D49F] [UDP:[fe80::8b:40d6:8fab:70f%en0]:5552] --- Type 0001:02 (IM:ReadRequest) (B:52)
[1729240534.533] [37163:743182] [EM] ??1 [E:41276i S:17665 M:258475850] (S) Msg Retransmission to 1:FFFFFFEFFFFFFFFF in 626ms [State:Idle II:500 AI:300 AT:4000]
[1729240534.533] [37163:743182] [DMG] MoveToState ReadClient[0x138f04080]: Moving to [AwaitingIn]
[1729240534.535] [37163:743182] [EM] >>> [E:41276i S:17665 M:146351207 (Ack:258475850)] (S) Msg RX from 1:FFFFFFEFFFFFFFFF [D49F] to C5AA883353531361 --- Type 0001:05 (IM:ReportData) (B:68)
[1729240534.535] [37163:743182] [EM] Found matching exchange: 41276i, Delegate: 0x138f04090
[1729240534.535] [37163:743182] [EM] Rxd Ack; Removing MessageCounter:258475850 from Retrans Table on exchange 41276i
[1729240534.535] [37163:743182] [DMG] ReportDataMessage =
[1729240534.535] [37163:743182] [DMG] {
[1729240534.535] [37163:743182] [DMG] 	AttributeReportIBs =
[1729240534.535] [37163:743182] [DMG] 	[
[1729240534.535] [37163:743182] [DMG] 		AttributeReportIB =
[1729240534.535] [37163:743182] [DMG] 		{
[1729240534.535] [37163:743182] [DMG] 			AttributeStatusIB =
[1729240534.535] [37163:743182] [DMG] 			{
[1729240534.535] [37163:743182] [DMG] 				AttributePathIB =
[1729240534.535] [37163:743182] [DMG] 				{
[1729240534.535] [37163:743182] [DMG] 					Endpoint = 0x4,
[1729240534.535] [37163:743182] [DMG] 					Cluster = 0x50d,
[1729240534.535] [37163:743182] [DMG] 					Attribute = 0x0000_0004,
[1729240534.535] [37163:743182] [DMG] 				}
[1729240534.535] [37163:743182] [DMG] 					
[1729240534.535] [37163:743182] [DMG] 				StatusIB =
[1729240534.535] [37163:743182] [DMG] 				{
[1729240534.535] [37163:743182] [DMG] 					status = 0x86 (UNSUPPORTED_ATTRIBUTE),
[1729240534.535] [37163:743182] [DMG] 				},
[1729240534.535] [37163:743182] [DMG] 				
[1729240534.535] [37163:743182] [DMG] 			},
[1729240534.535] [37163:743182] [DMG] 			
[1729240534.535] [37163:743182] [DMG] 		},
[1729240534.535] [37163:743182] [DMG] 		
[1729240534.535] [37163:743182] [DMG] 	],
[1729240534.535] [37163:743182] [DMG] 	
[1729240534.535] [37163:743182] [DMG] 	SuppressResponse = true, 
[1729240534.535] [37163:743182] [DMG] 	InteractionModelRevision = 11
[1729240534.535] [37163:743182] [DMG] }
[1729240534.535] [37163:743182] [TOO] Response Failure: IM Error 0x00000586: General error: 0x86 (UNSUPPORTED_ATTRIBUTE)
`

### Platform

android

### Platform Version(s)

1.3

### Type

Manually tested with SDK

nicelyjust, 2024-10-18:
the linux tv-casting-app against the linux tv-app had the same issue
`> cast cluster applicationbasic read application 0 4
[1729242277.649] [37163:746802] [TOO] Run command failure: examples/tv-casting-app/tv-casting-common/commands/common/CHIPCommand.cpp:74: CHIP Error 0x00000032: Timeout
Done
> [1729242277.649] [37163:746802] [TOO] nodeId set to 0, using default for fabric 1
[1729242277.649] [37163:746802] [-] Binding type=1 fab=1 nodeId=0xFFFFFFEFFFFFFFFF groupId=65535 local endpoint=0 remote endpoint=1 cluster=0x0000_0000
[1729242277.649] [37163:746802] [-] GetVideoPlayerNodeForFabricIndex nodeId=0xFFFFFFEFFFFFFFFF
[1729242277.649] [37163:746802] [TOO] Sending command to node 0xffffffefffffffff
[1729242277.649] [37163:746802] [CSM] FindOrEstablishSession: PeerId = [1:FFFFFFEFFFFFFFFF]
[1729242277.649] [37163:746802] [CSM] FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1729242277.649] [37163:746802] [DIS] Found an existing secure session to [1:FFFFFFEFFFFFFFFF]!
[1729242277.649] [37163:746802] [DIS] OperationalSessionSetup[1:FFFFFFEFFFFFFFFF]: State change 1 --> 5
[1729242277.649] [37163:746802] [TOO] ModelCommand::OnDeviceConnectedFn
[1729242277.649] [37163:746802] [TOO] Sending ReadAttribute to:
[1729242277.649] [37163:746802] [TOO] 	cluster 0x0000_050D, attribute: 0x0000_0004, endpoint 4
[1729242277.649] [37163:746802] [DMG] SendReadRequest ReadClient[0x138f04080]: Sending Read Request
[1729242277.649] [37163:746802] [DMG] 0 data version filters provided, 0 not relevant, 0 encoded, 0 skipped due to lack of space
[1729242277.649] [37163:746802] [EM] <<< [E:41278i S:17665 M:258475854] (S) Msg TX from C5AA883353531361 to 1:FFFFFFEFFFFFFFFF [D49F] [UDP:[fe80::8b:40d6:8fab:70f%en0]:5552] --- Type 0001:02 (IM:ReadRequest) (B:52)
[1729242277.650] [37163:746802] [EM] ??1 [E:41278i S:17665 M:258475854] (S) Msg Retransmission to 1:FFFFFFEFFFFFFFFF in 679ms [State:Idle II:500 AI:300 AT:4000]
[1729242277.650] [37163:746802] [DMG] MoveToState ReadClient[0x138f04080]: Moving to [AwaitingIn]
[1729242277.652] [37163:764298] [EM] >>> [E:41278i S:17665 M:146351209 (Ack:258475854)] (S) Msg RX from 1:FFFFFFEFFFFFFFFF [D49F] to C5AA883353531361 --- Type 0001:05 (IM:ReportData) (B:68)
[1729242277.652] [37163:764298] [EM] Found matching exchange: 41278i, Delegate: 0x138f04090
[1729242277.652] [37163:764298] [EM] Rxd Ack; Removing MessageCounter:258475854 from Retrans Table on exchange 41278i
[1729242277.652] [37163:764298] [DMG] ReportDataMessage =
[1729242277.652] [37163:764298] [DMG] {
[1729242277.652] [37163:764298] [DMG] 	AttributeReportIBs =
[1729242277.652] [37163:764298] [DMG] 	[
[1729242277.652] [37163:764298] [DMG] 		AttributeReportIB =
[1729242277.652] [37163:764298] [DMG] 		{
[1729242277.652] [37163:764298] [DMG] 			AttributeStatusIB =
[1729242277.652] [37163:764298] [DMG] 			{
[1729242277.652] [37163:764298] [DMG] 				AttributePathIB =
[1729242277.652] [37163:764298] [DMG] 				{
[1729242277.652] [37163:764298] [DMG] 					Endpoint = 0x4,
[1729242277.652] [37163:764298] [DMG] 					Cluster = 0x50d,
[1729242277.652] [37163:764298] [DMG] 					Attribute = 0x0000_0004,
[1729242277.652] [37163:764298] [DMG] 				}
[1729242277.652] [37163:764298] [DMG] 					
[1729242277.652] [37163:764298] [DMG] 				StatusIB =
[1729242277.652] [37163:764298] [DMG] 				{
[1729242277.652] [37163:764298] [DMG] 					status = 0x86 (UNSUPPORTED_ATTRIBUTE),
[1729242277.652] [37163:764298] [DMG] 				},
[1729242277.652] [37163:764298] [DMG] 				
[1729242277.652] [37163:764298] [DMG] 			},
[1729242277.652] [37163:764298] [DMG] 			
[1729242277.652] [37163:764298] [DMG] 		},
[1729242277.652] [37163:764298] [DMG] 		
[1729242277.652] [37163:764298] [DMG] 	],
[1729242277.652] [37163:764298] [DMG] 	
[1729242277.652] [37163:764298] [DMG] 	SuppressResponse = true, 
[1729242277.652] [37163:764298] [DMG] 	InteractionModelRevision = 11
[1729242277.652] [37163:764298] [DMG] }
[1729242277.652] [37163:764298] [TOO] Response Failure: IM Error 0x00000586: General error: 0x86 (UNSUPPORTED_ATTRIBUTE)
[1729242277.652] [37163:764298] [EM] <<< [E:41278i S:17665 M:258475855 (Ack:146351209)] (S) Msg TX from C5AA883353531361 to 1:FFFFFFEFFFFFFFFF [D49F] [UDP:[fe80::8b:40d6:8fab:70f%en0]:5552] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:34)
[1729242277.652] [37163:764298] [EM] Flushed pending ack for MessageCounter:146351209 on exchange 41278i`

chrisdecenzo, 2024-10-24:
It looks like something changed in the main SDK code which encodes attributes of type struct, and this broke the reading of the struct attributes in the ApplicationBasic cluster. I suspect that other media clusters with struct attributes would also be impacted.

The code which likely needs to be updated to whatever the new convention is can be found here:

src/app/clusters/application-basic-server/application-basic-server.cpp:

```
CHIP_ERROR Delegate::HandleGetApplication(app::AttributeValueEncoder & aEncoder)
{
    ApplicationBasicApplicationType application;
    application.catalogVendorID = mCatalogVendorApp.catalogVendorId;
    application.applicationID   = CharSpan(mCatalogVendorApp.applicationId, strlen(mCatalogVendorApp.applicationId));
    return aEncoder.Encode(application);
}
```

We should verify that this is getting called, and then double check that the values being assigned to application.catalogVendorID and application.applicationID are valid.

nicelyjust, 2024-11-14:
the HandleGetApplication is not getting called @chrisdecenzo

nicelyjust, 2024-11-14:
There is another issue where the feature-map attribute cannot be retrieved, and a generic error code is returned.