project-chip/connectedhomeip #36142: [BUG]DoorLock subscribeAttributeLockState is not working for Lock state

muralivm, 2024-10-18:
### Reproduction steps

I am fetching the matter accessory from Apple Home to my app using apple home APIs
I am able to read and write the attributes to Matter accessory using Matter Spec. 

Commission the Matter Accessory to Apple Home
Fetch the Matter accessory to iOS APP
Able to read and write the attributes for Matter accessory using Matter spec. 
But DoorLock subscribeAttributeLockState is not working 

Please find the custom code 
 doorLockCluster.subscribeAttributeLockState(with: MTRSubscribeParams(minInterval: NSNumber(value: 1), maxInterval: NSNumber(value: 2)), subscriptionEstablished: {
            print("subscriptionEstablished")
        }, reportHandler: { val, errro in
            print("reportHandler")
            print(errro)
        })

I am unable to find the documentation for this API as well, the subscription is not working for the Lock state. 

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

2574847dad7c1042e649241c3c39dd48a5d5f5b5

### Platform

darwin

### Anything else?

I am fetching the matter accessory from Apple Home to my app using apple home APIs
I am able to read and write the attributes to Matter accessory using Matter Spec. 

Commission the Matter Accessory to Apple Home
Fetch the Matter accessory to iOS APP
Able to read and write the attributes for Matter accessory using Matter spec. 
But DoorLock subscribeAttributeLockState is not working 

Please find the custom code 
 doorLockCluster.subscribeAttributeLockState(with: MTRSubscribeParams(minInterval: NSNumber(value: 1), maxInterval: NSNumber(value: 2)), subscriptionEstablished: {
            print("subscriptionEstablished")
        }, reportHandler: { val, errro in
            print("reportHandler")
            print(errro)
        })

I am unable to find the documentation for this API as well, the subscription is not working for the Lock state.

bzbarsky-apple, 2024-10-18:
> I am fetching the matter accessory from Apple Home to my app using apple home APIs

How _exactly_ are you doing this?

I suggest following the directions at https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#providing-feedback-to-apple and putting the feedback assistant ticket ID in this issue.  I'll prod people to look at it.

muralivm, 2024-10-21:
Do you have document for the subscribe state for fetching Accessory from Apple Home ?

bzbarsky-apple, 2024-10-31:
@muralivm I am not sure what you are asking, to be honest.

woody-apple, 2025-04-02:
Please file a feedback ticket here:
https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#providing-feedback-to-apple

closed 2025-04-02 by woody-apple