project-chip/connectedhomeip #36146: [BUG] Not able to run python scripts

rquidute, 2024-10-18:
### Reproduction steps

I'm not able to run python scripts.
Command: `python3 /root/python_testing/scripts/sdk/TC_ACE_1_3.py --commissioning-method on-network --discriminator 3840 --passcode 20202021`

Getting this error: 
```
Traceback (most recent call last):
  File "/root/python_testing/scripts/sdk/TC_ACE_1_3.py", line 41, in <module>
    from chip.testing.matter_testing import MatterBaseTest, TestStep, async_test_body, default_matter_test_main
ModuleNotFoundError: No module named 'chip.testing'
```

So, trying some code changes by editing the import at TC_ACE_1_3.py from `from chip.testing.matter_testing import MatterBaseTest, TestStep, async_test_body, default_matter_test_main`
 to `from matter_testing_infrastructure.chip.testing.matter_testing import MatterBaseTest, TestStep, async_test_body, default_matter_test_main`
 and now I'm getting another error:
```
 Traceback (most recent call last):
  File "/root/python_testing/scripts/sdk/TC_ACE_1_3.py", line 41, in <module>
    from matter_testing_infrastructure.chip.testing.matter_testing import MatterBaseTest, TestStep, async_test_body, default_matter_test_main
  File "/root/python_testing/scripts/sdk/matter_testing_infrastructure/chip/testing/matter_testing.py", line 70, in <module>
    from chip.testing.global_attribute_ids import GlobalAttributeIds
ModuleNotFoundError: No module named 'chip.testing'
```

### Bug prevalence

always

### GitHub hash of the SDK that was being used

0c59ebf6b37cd16f196db42e43f5f6269727db6c

### Platform

python

bzbarsky-apple, 2024-10-18:
@cecille

andy31415, 2024-10-24:
You will need to install the `whl` packages to run tests.

`./scripts/build_python.sh -i out/venv` would install a virtualenv in `out/venv` for example.
In particular you need to compile the target `//src/python_testing/matter_testing_infrastructure:chip-testing.wheel` and then install the wheel (https://github.com/project-chip/connectedhomeip/blob/master/src/python_testing/matter_testing_infrastructure/BUILD.gn#L21)

In practice, use `build_python.sh` to have an environment set up for you.

andy31415, 2024-10-24:
This is how python tests are run in CI: https://github.com/project-chip/connectedhomeip/blob/master/.github/workflows/tests.yaml#L524

Notice the usage of `run_in_python_env.sh out/venv`. For your manual runs you will have to do a `source out/venv/bin/activate.sh` or similar. Note that this switches the virtualenv from any bootstrapped environment.

Please also do not install the packages in your bootstrap environment - we do not support that as it is too easy to cause hard to debug references to old files while building.

closed 2024-10-24 by andy31415