project-chip/connectedhomeip #36214: [VSCode ] Failed to open devcontainer

FergusHuang, 2024-10-23:
### Reproduction steps

VSCode failed to open the latest master version project in the devcontainer.
[remoteContainers-2024-10-23T03-17-51.605Z.log](https://github.com/user-attachments/files/17491779/remoteContainers-2024-10-23T03-17-51.605Z.log)

### Platform

vscode

### Platform Version(s)

master

### Type

Manually tested with SDK

### Anything else?

Error log: groupadd: GID '1000' already exists
I checked the base image(chip-build-vscode) for tag 74, and it does contain the group with GID 1000. However, the image for tag 22 does not include this group. So that is the reason that triggered the issue.

andy31415, 2024-10-24:
@FergusHuang could you create a PR to test this? 

This seems to be from https://github.com/project-chip/connectedhomeip/blob/master/.devcontainer/Dockerfile#L43 so if these user/groups already exist we could increase their IDs at https://github.com/project-chip/connectedhomeip/blob/master/.devcontainer/Dockerfile#L22

FergusHuang, 2024-10-25:
I tried to change UID to 1001 by passing uid option to build.sh. But that will bring some permission issues.
First of all, for the folders like out/ or .devcontainer/ which need to be written by VSCode, these folders were created by UID 1000, so UID 1001 will fail to write to them. We can at least add UID 1001 to GID 1000 to solve this issue.
Hower, in another case  I found some files only have READ permission for the owner group is  as below:
-rw-r--r-- 1 ubuntu ubuntu 856 Aug  2 12:29 /workspaces/connectedhomeip/build_overrides/pigweed_environment.gni
This kind of situation is more annoying because you have to check all the files in the project.
In summary, I believe the simplest solution is to remove GID/UID 1000 from the base image.

PR #36528 (2024-11-19): VSCode docker image: Delete user and group in case of conflicts
Fixes #36214 

The build of the DevContainer fails if the local UID:GID is 1000:1000, as normally happens in a personal machine, because the Dockerfile try to add a user and a group that already exists in the base image.

This PR proposes a simple, yet not perfect solution: if a conflicting UID or GID exists in the base Docker image, the existing entities are deleted before creating the new ones.
closed 2024-11-19 by mergify[bot]