project-chip/connectedhomeip #36262: [1.2] Homekit device not Supported IoS 18.1

mertmzk, 2024-10-28:
### Reproduction steps

Hello,
I have run the RVC example on Smarthings hub successfully but when I try to pair it with Homekit it is paring but showing as device not supported. 

I have installed the certificates to both on Iphone and Homepod, but still no luck.

Any suggestions?

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

v1.2.0.1

### Platform

raspi

### Type

Platform Issue

bzbarsky-apple, 2024-10-28:
@mertmzk Please follow the directions at https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#providing-feedback-to-apple to file a Feedback Assistant issue.  Mention the issue ID here and I can ask folks to take a look.

Though note that I believe RVCs are not yet a supported device type in Apple Home on iOS 18.1, which might be all that's going on here....

mertmzk, 2024-10-29:
@bzbarsky-apple  Release note from IoS 18 was showing it will support matter RVCs.

Anyway created Feedback assistant issue and issue Id is : FB15622701

bzbarsky-apple, 2024-10-29:
> Release note from IoS 18 was showing it will support matter RVCs.

What exact release note are you looking at?

I checked carefully, and 18.0 and 18.1 do not support Matter RVCs out of the box yet.

closed 2024-10-29 by bzbarsky-apple