project-chip/connectedhomeip #36333: [1.2] Set curtain percentage in Homekit home, curtain UI is constantly updated

wolen666, 2024-11-01:
### Reproduction steps

Here are my Zap settings and demonstration video，I found that clicking on the percentage inside the curtain has already been updated, but the external card has been constantly updated.I am using the example of the light switch app to add curtain accessories directly to Zap

https://github.com/user-attachments/assets/99e84e6e-bb3a-427c-9452-e886849e6442

[image]
[image]
[image]

### Bug prevalence

1-2 times week

### GitHub hash of the SDK that was being used

a9027ca46e915adac434536b11037b549d83b8d9

### Platform

esp32

### Platform Version(s)

1.3

### Type

Platform Issue

bzbarsky-apple, 2024-11-01:
Please follow the directions at https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#providing-feedback-to-apple to report the issue to Apple, with logs that will show what the Home side is seeing.  Mention the feedback ticket ID here, and I'll prod people to take a look.

woody-apple, 2025-04-02:
Please file a feedback ticket here.

closed 2025-04-02 by woody-apple