project-chip/connectedhomeip #36346: [BUG] Reinitializing ChipController on Android faults in ICDClientStorage

jonsmirl, 2024-11-02:
### Reproduction steps

@yunhanw-google 
My Android app supports multiple homes, when I switch homes I need to make another ChipController object because I have to initialize using a different operational key. Now when I make the second controller I am faulting in the ICD code. I have no ICD specific code in my Android app. I don't explicitly delete the first controller instance, I was relying on garbage collection to recover it so there will be two controller instances around until that happens. This is code which had been working for several years.  Do I have to do something special now for ICD support? 

Note that I am a little bit behind head (Aug 27), if this has been fixed I can update. But I looked in the git logs and didn't see anything that looked like a fix.
~~~
2024-11-01 23:33:00.500 30717-30717 AndroidRuntime          com.lowpan.m2                        E  FATAL EXCEPTION: main (Ask Gemini)
                                                                                                    Process: com.lowpan.m2, PID: 30717
                                                                                                    chip.devicecontroller.ChipDeviceControllerException: src/app/icd/client/DefaultICDClientStorage.cpp:159: CHIP Error 0x00000003: Incorrect state
                                                                                                    	at chip.devicecontroller.ChipDeviceController.newDeviceController(Native Method)
                                                                                                    	at chip.devicecontroller.ChipDeviceController.<init>(ChipDeviceController.java:69)
                                                                                                    	at com.lowpan.m2.chip.ChipClient$1$1.emit(ChipClient.kt:196)
                                                                                                    	at com.lowpan.m2.chip.ChipClient$1$1.emit(ChipClient.kt:96)
                                                                                                    	at kotlinx.coroutines.flow.StateFlowImpl.collect(StateFlow.kt:396)
                                                                                                    	at kotlinx.coroutines.flow.StateFlowImpl$collect$1.invokeSuspend(Unknown Source:15)
                                                                                                    	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
                                                                                                    	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:108)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:959)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:100)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:232)
                                                                                                    	at android.os.Looper.loop(Looper.java:317)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8705)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:580)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:886)
~~~

~~~
                    chipDeviceController = ChipDeviceController(
                        ControllerParams.newBuilder(operationalKeyConfig()).setUdpListenPort(0)
                            .setControllerVendorId(Lowpan.COMPANY).build()
                    ).also { chipDeviceController ->
~~~

### Bug prevalence

always

### GitHub hash of the SDK that was being used

ea679d2dc674f576f4d391d1d71af1489010e580

### Platform

android

yunhanw-google, 2024-11-03:
@jonsmirl  thanks for pointing out this issue, could you have a try on this fix?
https://github.com/project-chip/connectedhomeip/pull/36348

jonsmirl, 2024-11-03:
I tested in my app and fault is gone. I am not using ICD so I don't know if that was still working.

yunhanw-google, 2024-11-13:
@jonsmirl I just merged the fix, please retest when you have time, thanks

CaoJieNan, 2024-12-10:
I saw commit #36348 on the master branch, which fixes this issue. I pulled the code from the matter branch, recompiled it into an android-library, and then tried to create two "ChipDeviceController". However, I still get this error: "src/app/icd/client/DefaultICDClientStorage.cpp:164: CHIP Error 0x00000003: Incorrect state".

[image]

CaoJieNan, 2024-12-10:
I added some test logs in "AndroidDeviceControllerWrapper.cpp" to verify that the android-library I compiled is working. After checking in Android Studio, I confirmed that the test logs were output. I'm confused because the function "getICDClientStorage()->Shutdown()" was called, but it still reports an error: "src/app/icd/client/DefaultICDClientStorage.cpp:164: CHIP Error 0x00000003: Incorrect state".

[image]
[image]

marcokhoo, 2024-12-10:
Our app is seeing the same problem in Andriod.  iOS is working well. We tried to switch between the control of 2 different fabrics ( which supporting icd device)by reinitiating the matter chip controller.Please recommend for work around or help to fix this problem . Thx

CaoJieNan, 2024-12-12:
I’ve found a solution to the issue. I implemented a custom method named deleteDeviceControllerCustom in the source code. By invoking deleteDeviceControllerCustom every time a new ChipDeviceController is created, the previous instance of ChipDeviceController is properly destroyed. This prevents the occurrence of the error CHIP Error 0x00000003: Incorrect state.

[image]

[image]

yunhanw-google, 2025-02-04:
@CaoJieNan @marcokhoo
Could you validate this PR? 
https://github.com/project-chip/connectedhomeip/pull/37396

Thanks

CaoJieNan, 2025-02-05:
@yunhanw-google Thank you very much for fixed this issue, I'll verify this PR later!

closed 2025-02-05 by yunhanw-google