project-chip/connectedhomeip #36352: [BUG] Crash after OTA successfully finished

MrLiuYunPing, 2024-11-04:
### Reproduction steps

OTA device, after finished, the app will crash.

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

branch: master; commit id is:93abcaed41239c2a4ad675de30dfde953dcfd5c1

### Platform

android

### Platform Version(s)

android 14

### Anything else?

[crash.log](https://github.com/user-attachments/files/17616975/crash.log)

MrLiuYunPing, 2024-11-05:
Call finishOTAProvider() will cause the crash.

MrLiuYunPing, 2024-11-05:
src/controller/java/BdxOTASender.cpp
void BdxOTASender::ResetState()
`
Responder::ResetTransfer(); `should change to 
`ResetTransfer();`

MrLiuYunPing, 2024-11-05:
fix: drop poll timer to avoid crash after TransferFacilitator destruction (https://github.com/project-chip/connectedhomeip/pull/34538) Ithink, this fix causes this crash.

andy31415, 2024-11-25:
@MrLiuYunPing should we mark this as closed after 34538?

MrLiuYunPing, 2024-11-26:
> @MrLiuYunPing should we mark this as closed after 34538?

No, I mean 34538 caused the crash, because before this change, it won't crash.

MrLiuYunPing, 2024-11-27:
/src/protocols/bdx/TransferFacilitator.cpp:
`void TransferFacilitator::ResetTransfer()
{
    mTransfer.Reset();
    ChipLogProgress(BDX, "Stop polling for messages");

    VerifyOrReturn(mSystemLayer != nullptr);
    mSystemLayer->CancelTimer(PollTimerHandler, this);
    mSystemLayer = nullptr;//add this line
}`

/src/controller/java/BdxOTASender.cpp

`CHIP_ERROR BdxOTASender::Shutdown()
{
    assertChipStackLockedByCurrentThread();

//    VerifyOrReturnError(mSystemLayer != nullptr, CHIP_ERROR_INCORRECT_STATE);
    VerifyOrReturnError(mExchangeMgr != nullptr, CHIP_ERROR_INCORRECT_STATE);

    mExchangeMgr->UnregisterUnsolicitedMessageHandlerForProtocol(Protocols::BDX::Id);
    ResetState();

    mExchangeMgr = nullptr;
//    mSystemLayer = nullptr;

    return CHIP_NO_ERROR;
}`

yunhanw-google, 2024-12-05:
@MrLiuYunPing  I think the fix is correct, I just create the PR for this issue, btw, may I know your slack? you may reach to me([Google]yunhanw) directly if there is further android matter issue on it in case we miss the issue in github. thanks

closed 2024-12-06 by yunhanw-google

MrLiuYunPing, 2024-12-09:
> CancelTimer
Sorry, I don't have an account on Slack.
CancelTimer() won't work when PollTimerHandler has been called, and then PollForOutput will be called again. mSystemLayer will call StartTimer again. Then HandleTransferSessionOutput will be called form a deallocated object. The android app will crash.

yunhanw-google, 2024-12-11:
@MrLiuYunPing Do you have wechat?