project-chip/connectedhomeip #36353: [BUG] Attribute change cannot be reported for the data model provider which doesn't use ember

wqx6, 2024-11-04:
### Reproduction steps

For the attributes that are managed by the AttributeAccessInterface, when their changes are not caused by the WriteAttribute function in the data model provider, these changes cannot be reported if the Data Model provider doesn't use ember.

For instance, `LastNetworkingStatus` in `NetworkCommissioning` cluster will be changed after network connection changes but the `WriteAttribute` will not be called. `MatterReportingAttributeChangeCallback()` will be called instead. 

Should we provide an interface `ReportAttributeChanged` in data model provider and call that function in `MatterReportingAttributeChangeCallback()`?

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

744f45cc886210ee95404ea659fd0327f818e891

### Platform

core

bzbarsky-apple, 2024-11-04:
@andy31415

andy31415, 2024-11-04:
The path build for this in `DataModel::Provider` is to use a https://github.com/project-chip/connectedhomeip/blob/master/src/app/data-model-provider/ProviderChangeListener.h#L42 (which is mapped by the InteractionModelEngine to report attribute changes). 

The way this gets passed is that a https://github.com/project-chip/connectedhomeip/blob/master/src/app/data-model-provider/Context.h#L32 contains a ProviderChangeListener and this context is passed to the Provider init and is accessible in GetCurrentContext https://github.com/project-chip/connectedhomeip/blob/master/src/app/data-model-provider/Provider.h#L60

The bindings however mean that the Provider is currently not piped through into the cluster implementations. Once could use `InteractionModelEngine::GetInstance().GetDataModelProvider()` however that is a bit coupled.

So for the long term, what we need is AAI to be aware of the context (passed in) or maybe define a "cluster implementation" that actually has all data and combines AAI and CHI because this is what people really want (see examples/tv-casting-app/tv-casting-common/core/BaseCluster.h having done this or https://github.com/project-chip/connectedhomeip/pull/33480 that wanted to propose a base class)

This is what I have on my medium TODO list:
  - want to have something that combines AAI + CHI + metadata and context access
  - have clusters implement/use that instead of what we have today
  
No short-term fix though, except maybe `InteractionModelEngine::GetInstance()->GetDataModelProvider()->CurrentContext().dataModelChangeListener->MarkDirty(attributePath)` (eek at how long that is!)

wqx6, 2024-11-05:
@andy31415 Could this #36364  be a short term fix for this issue?

andy31415, 2024-11-05:
> @andy31415 Could this #36364 be a short term fix for this issue?

I added a changes requested to that one. My thinking on the data dependencies is that you do not own the datamodel::provider data, the provider owns it. So it should not be possible to `"allow external code to tell the data model provider its own data was changed"` because the external code should now own the data.

What we could do is attach a ProviderChangeListener (or maybe even context) to the AttributeAccessInterface / CommandHandlerInterface and allow that to be called. That listener should pipe through both the provider itself (to increase versions) and then pipe through the main listener (to trigger reporting engine updates).