project-chip/connectedhomeip #36417: [1.4]  In WaterHeaterManagement Cluster, The Featuremap value is Failure instead of Success.

Ashwinigrl, 2024-11-07:
### Reproduction steps

1. Execute the below mentioned command to put DUT into a commissionable state :
`./chip-energy-management-app`

2. Once DUT reach the commissionable state sent below mentioned command on TH.
`./chip-tool pairing onnetwork 1 20202021`

Once after successful pairing provide the below commands:

1. TH reads FeatureMap value from the DUT.

` ./chip-tool waterheatermanagement read feature-map 1 1`

**Actual Outcome:**

> [1730974880.590] [65803:65805] [DMG] 				{
> [1730974880.590] [65803:65805] [DMG] 					status = 0x01 (FAILURE),
> [1730974880.590] [65803:65805] [DMG] 				},
> [1730974880.590] [65803:65805] [DMG] 				
> [1730974880.590] [65803:65805] [DMG] 			},
> [1730974880.590] [65803:65805] [DMG] 			
> [1730974880.590] [65803:65805] [DMG] 		},
> [1730974880.590] [65803:65805] [DMG] 		
> [1730974880.590] [65803:65805] [DMG] 	],
> [1730974880.590] [65803:65805] [DMG] 	
> [1730974880.590] [65803:65805] [DMG] 	SuppressResponse = true, 
> [1730974880.590] [65803:65805] [DMG] 	InteractionModelRevision = 11
> [1730974880.590] [65803:65805] [DMG] }

**Expected Outcome:**

```
[1730975412.558] [65847:65849] [DMG] 				Data = 3 (unsigned), 
[1730975412.558] [65847:65849] [DMG] 			},
[1730975412.558] [65847:65849] [DMG] 			
[1730975412.558] [65847:65849] [DMG] 		},
[1730975412.558] [65847:65849] [DMG] 		
[1730975412.558] [65847:65849] [DMG] 	],
[1730975412.558] [65847:65849] [DMG] 	
[1730975412.558] [65847:65849] [DMG] 	SuppressResponse = true, 
[1730975412.558] [65847:65849] [DMG] 	InteractionModelRevision = 11
[1730975412.558] [65847:65849] [DMG] }
[1730975412.558] [65847:65849] [TOO] Endpoint: 1 Cluster: 0x0000_0094 Attribute 0x0000_FFFC DataVersion: 3823610553
[1730975412.559] [65847:65849] [TOO]   FeatureMap: 3
```

In the WaterHeaterManagement Cluster, When we read the FeatureMap attribute from the DUT returns a failure response instead of a success response with the expected FeatureMap value. The sample app(Energy-Management-app) needs to be updated to correct the FeatureMap attribute response.

**Sample App used to read the FeatureMap value:**  `EnergyManagementApp`

### Bug prevalence

Whenever we do

### GitHub hash of the SDK that was being used

9c389ba953caa435d1f1ed34d5b799e9d1a7a4ea

### Platform

raspi

### Platform Version(s)

1.4

### Type

Test Improvement

### Anything else?

**Spec References:**

- https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/WaterHeaterManagement.adoc#4-features

**Log References:**
[WaterHeaterManagement.txt](https://github.com/user-attachments/files/17660074/WaterHeaterManagement.txt)

bzbarsky-apple, 2024-11-09:
@jamesharrow

lboue, 2024-11-11:
Did you try against PR #36201?

jamesharrow, 2024-11-11:
The issue is that you need to use `--application water-heater` to enable the Water Heater cluster to respond on EP1 on Linux (or compile this for an embedded target with water heater enabled).

@ashwinigrl - please retest with the extra command line option.

Note that the PR #36201 will move the Water Heater to a different endpoint once merged.

jamesharrow, 2025-05-20:
I think this PR should be closed now. The water heater has moved to a separate endpoint based on the above PR that was merged several months ago.

Closing the issue.

closed 2025-05-20 by jamesharrow