project-chip/connectedhomeip #36420: [BUG] Invalid command in `WriteAcl` blocks access from devices by deleting all ACL entries.

BoB13-Matter, 2024-11-07:
### Reproduction steps

Send an invalid privilege value in a command to WriteAcl
```
./chip-tool accesscontrol write acl '[{"privilege":99,"authMode":2,"subjects":[112233],"targets":[],"fabricIndex":1}]' 1 0
``` 

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

all version

### Platform

core

### Anything else?

## **Summary**

- **Security Concern**: Invalid command in `WriteAcl` blocks access from devices by deleting all ACL entries.

## **Description**

When an invalid command is provided to the `WriteAcl` function in the Access Control Cluster, it may lead to the deletion of all existing Access Control List (ACL) entries, blocking access not only from devices within the same fabric but also from administrators.

- **Vulnerability Details**: The `WriteAcl` function is designed to delete existing ACL entries before creating new ones based on the provided input. However, if an invalid input is detected, the function fails at the decoding stage, triggering a `CONSTRAINT_ERROR` that halts the process before any new entries are established.
- **Security Implications**: This issue isolates the affected device entirely, preventing all devices within the same fabric—and even administrators—from accessing it. In environments where secure and consistent access to each device is critical, this vulnerability could create significant operational risks by leaving the device unmanaged and disconnected. The loss of administrative access further escalates the issue, as it hinders any immediate corrective actions or oversight, thereby compromising the stability and reliability of systems dependent on this device.
- **Recommended Solution**: To prevent this issue, input decoding should be conducted at the beginning of the `WriteAcl` function. By validating the input before any deletion occurs, existing ACL entries can be preserved until the new entries are confirmed to be valid. This solution safeguards secure access for both administrators and other devices within the fabric, ensuring continuity in device management and network stability.

## Error Log
Returns `CONSTRAINT_ERROR`
```
>>> accesscontrol write acl '[{"privilege":99,"authMode":2,"subjects":[112233],"targets":[],"fabricIndex":1}]' 1 0
[1730957036.169] [106681:106681] [TOO] Command: accesscontrol write acl [{"privilege":99,"authMode":2,"subjects":[112233],"targets":[],"fabricIndex":1}] 1 0
[1730957036.169] [106681:106683] [TOO] Sending command to node 0x1
[1730957036.170] [106681:106683] [CSM] FindOrEstablishSession: PeerId = [1:0000000000000001]
[1730957036.170] [106681:106683] [CSM] FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1730957036.170] [106681:106683] [DIS] Found an existing secure session to [1:0000000000000001]!
[1730957036.170] [106681:106683] [DIS] OperationalSessionSetup[1:0000000000000001]: State change 1 --> 5
[1730957036.170] [106681:106683] [TOO]  cluster 0x0000_001F, attribute: 0x0000_0000, endpoint 0
[1730957036.170] [106681:106683] [DMG] WriteClient moving to [AddAttribu]
[1730957036.170] [106681:106683] [DMG] WriteClient moving to [AddAttribu]
[1730957036.170] [106681:106683] [EM] <<< [E:6116i S:40323 M:211704883] (S) Msg TX from 000000000001B669 to 1:0000000000000001 [5922] [UDP:[fe80::8815:299d:a000:5ccd%wlp3s0]:5540] --- Type 0001:06 (IM:WriteRequest) (B:97)
[1730957036.170] [106681:106683] [EM] ??1 [E:6116i S:40323 M:211704883] (S) Msg Retransmission to 1:0000000000000001 in 621ms [State:Idle II:500 AI:300 AT:4000]
[1730957036.170] [106681:106683] [DMG] WriteClient moving to [AwaitingRe]
[1730957036.179] [106681:106683] [EM] >>> [E:6116i S:40323 M:248075942 (Ack:211704883)] (S) Msg RX from 1:0000000000000001 [5922] to 000000000001B669 --- Type 0001:07 (IM:WriteResponse) (B:84)
[1730957036.179] [106681:106683] [EM] Found matching exchange: 6116i, Delegate: 0x7d99c803a610
[1730957036.179] [106681:106683] [EM] Rxd Ack; Removing MessageCounter:211704883 from Retrans Table on exchange 6116i
[1730957036.179] [106681:106683] [DMG] WriteClient moving to [ResponseRe]
[1730957036.179] [106681:106683] [DMG] WriteResponseMessage =
[1730957036.179] [106681:106683] [DMG] {
[1730957036.179] [106681:106683] [DMG]  AttributeStatusIBs =
[1730957036.179] [106681:106683] [DMG]  [
[1730957036.179] [106681:106683] [DMG]          AttributeStatusIB =
[1730957036.179] [106681:106683] [DMG]          {
[1730957036.179] [106681:106683] [DMG]                  AttributePathIB =
[1730957036.179] [106681:106683] [DMG]                  {
[1730957036.179] [106681:106683] [DMG]                          Endpoint = 0x0,
[1730957036.179] [106681:106683] [DMG]                          Cluster = 0x1f,
[1730957036.179] [106681:106683] [DMG]                          Attribute = 0x0000_0000,
[1730957036.179] [106681:106683] [DMG]                  }
[1730957036.179] [106681:106683] [DMG]
[1730957036.179] [106681:106683] [DMG]                  StatusIB =
[1730957036.179] [106681:106683] [DMG]                  {
[1730957036.179] [106681:106683] [DMG]                          status = 0x00 (SUCCESS),
[1730957036.179] [106681:106683] [DMG]                  },
[... 59 lines cut ...]
[1730957175.065] [106681:106683] [DMG]                          {
[1730957175.065] [106681:106683] [DMG]                                  Endpoint = 0x0,
[1730957175.065] [106681:106683] [DMG]                                  Cluster = 0x28,
[1730957175.065] [106681:106683] [DMG]                                  Attribute = 0x0000_0001,
[1730957175.065] [106681:106683] [DMG]                          }
[1730957175.065] [106681:106683] [DMG]
[1730957175.065] [106681:106683] [DMG]                          StatusIB =
[1730957175.065] [106681:106683] [DMG]                          {
[1730957175.065] [106681:106683] [DMG]                                  status = 0x7e (UNSUPPORTED_ACCESS),
[1730957175.065] [106681:106683] [DMG]                          },
[1730957175.065] [106681:106683] [DMG]
[1730957175.065] [106681:106683] [DMG]                  },
[1730957175.065] [106681:106683] [DMG]
[1730957175.065] [106681:106683] [DMG]          },
[1730957175.065] [106681:106683] [DMG]
[1730957175.065] [106681:106683] [DMG]  ],
[1730957175.065] [106681:106683] [DMG]
[1730957175.065] [106681:106683] [DMG]  SuppressResponse = true,
[1730957175.065] [106681:106683] [DMG]  InteractionModelRevision = 11
[1730957175.065] [106681:106683] [DMG] }
[1730957175.065] [106681:106683] [TOO] Response Failure: IM Error 0x0000057E: General error: 0x7e (UNSUPPORTED_ACCESS)
[1730957175.065] [106681:106683] [TOO] Run command failure: IM Error 0x0000057E: General error: 0x7e (UNSUPPORTED_ACCESS)
[1730957175.065] [106681:106683] [EM] <<< [E:6118i S:40323 M:211704888 (Ack:248075944)] (S) Msg TX from 000000000001B669 to 1:0000000000000001 [5922] [UDP:[fe80::8815:299d:a000:5ccd%wlp3s0]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:34)
[1730957175.065] [106681:106683] [EM] Flushed pending ack for MessageCounter:248075944 on exchange 6118i
```

bzbarsky-apple, 2024-11-09:
Input decoding cannot be conducted "at the beginning", because the wire encoding replaces the list with an empty list, then starts adding entries to it, no?

This is explicitly called out in the spec: the first entry in the list _must_ be a valid entry granting administrative privilege to some entity, otherwise you lose access to the device.

closed 2024-11-09 by bzbarsky-apple

BoB13-Matter, 2024-11-11:
> Input decoding cannot be conducted "at the beginning", because the wire encoding replaces the list with an empty list, then starts adding entries to it, no?
>
> This is explicitly called out in the [...]

Thank you for your response. 
To ensure I fully understand the intended behavior, could you please point me to the exact section in the specification where it states that "the first entry in the list must be a valid entry granting administrative privilege"? 
I reviewed the document but wasn’t able to find this particular requirement, so any guidance would be greatly appreciated.

bzbarsky-apple, 2024-11-15:
@BoB13-Matter Hmm... I am not seeing that in the current version of the spec.  I have not been following the spec closely, so it's possible the guidance was removed or rephrased...