project-chip/connectedhomeip #36426: Controlling the expanding number of device endpoints by allowing them to be combined

jonsmirl, 2024-11-07:
### Feature description

The reason people are asking for cluster extensions outside the spec https://github.com/project-chip/connectedhomeip/issues/36408 is because the existing Matter Light Switch device type is insanely hard to use. As I've discussed with Boris my light switch keypad device has become a Frankenstein monster after being forced to implement 42 different endpoints from a single device if it is going to do behave as the current spec requires.

There are two problems with lighting devices, one simple and one complex.

The simple one. In multi-gang devices sometimes you don't use all of the gangs. If the gang is not connected to a load you don't want to display a UI for it. The On/Off Cluster needs an attribute for '"NoLoad". Then there is a toggle in the phone app UI to suppress the display of devices with no load attached. This needs to be a toggle in the phone app since you need to see the NoLoad devices during setup, but once they are set up you want to suppress their display.

The complex one. The number of endpoints needed to construct a smart light switch is very high, and for my devices it is currently five endpoints per lighting load. When ganged devices are involved that number multiplies. My eight button keypads now have 42 endpoints -- Root, Aggregator + 8 * (Light Bulb, Light Switch, Generic Switch, Indicator Led, Electrical Power). Since there is no defined smart switch type those endpoints have to be grouped using the TAGLIST feature on the Descriptor cluster (Matter 1.3) or the Fixed Label cluster (pre-1.3). None of the Google/Apple/Amazon apps support this crazy collection of endpoints.

An exploding number of endpoints is a particularly acute problem with lighting. I have a very large Insteon installation which has 63 light switches and I am unable to replace it using Matter devices. If each of those 63 light switches has five endpoints, that's 315 endpoints to deal with and that becomes a major problem to manage. Light switches really should be a single endpoint just to keep things manageable in large installations.

I would propose a mechanism to help constrain this explosion of endpoints, initially explained here: https://github.com/project-chip/connectedhomeip/issues/36329 The proposal is that if the clusters needed to implement a device type are mutually exclusive and those endpoints are related, then you can combine those device types onto a single endpoint and list all of the device types. The purpose in doing this is not as much to reduce the number of endpoints (which does help) but instead to make the grouping obvious in ganged devices.

For example: Light Bulb, Light Switch, Generic Switch and Electrical Power are all mutually exclusive clusters and are related together to form a smart switch. It would be possible to combine those four endpoints into a single endpoint and list all four device types. This is backwards compatible because the device types are mutually exclusive, if older code uses the Light Bulb device type it will just ignore the other clusters.

If this were an option, my keypads would have 10 endpoints instead of 42 - Root, Aggregator + 8 * (combo Bulb/Switch/GS/EP). The need for a special indicator LED endpoint disappears if I have the "NoLoad" flag on the OnOff cluster.

Two years ago I used this illustration in an issue about trying to build a pair of light switches which could control the top and bottom of a staircase. Despite two years of effort I have still not been able to achieve this. We are very close now, but still not quite there. This illustrates the use of "NoLoad". One switch in the pair has the load attached and the other one doesn't but you still want the indicator LED to work. However, you don't want two Light controls in the UI because there aren't two light bulbs.

[image]

### Platform

all

bzbarsky-apple, 2024-11-09:
This is a spec issue, not an SDK issue.

jonsmirl, 2024-11-09:
This is the only place I can post, I am blocked from directly making spec comments. Seems to me that it should be free to comment on the spec and then pay for the privilege of voting to accept those comments.