project-chip/connectedhomeip #36430: [Build] No Native Executable Generated When Building for Android

Shangeev-AP, 2024-11-08:
### Build issue(s)

I am experiencing an issue when building ChipTool for Android. After following the steps to build the project, only the APK is generated, and the expected native executable is not present.

I have followed the steps mentioned in this documentation :
 https://project-chip.github.io/connectedhomeip-doc/platforms/android/android_building.html

Also, I have followed building from scripts, not from Android Studio.

### Platform

android

andy31415, 2024-11-25:
@Shangeev-AP I would expect APK builds when building android binaries. What is the `expected native executable` you mentioned and based on what documentation? could you provide a reference (not only documentation link, you have provided that, but the exact like/statement where it describes something should be there but it is not)?

We need to determine if some documentation is out of date or if the build is broken or something else. If an APK is created that does sound like at least some expected builds are happening.

andy31415, 2024-11-25:
For reference, our build system will not build a `chip-tool` command line binary for Android as far as I am aware.

Shangeev-AP, 2024-11-27:
> @Shangeev-AP I would expect APK builds when building android binaries. What is the `expected native executable` you mentioned and based on what documentation? could you provide a reference (not only [...]

For linux build,  it is building a chip-tool native executable. For Android, it is not mentioned anywhere about 'native executable' as for android building only .apk is generated.

I need a command line chip-tool executable after android build also. Is it possible?

**My Final Requirement :**
Currently only .apk is generated after building for chip-tool android, I expect a chip-tool arm-32 bit command line binary which can be executed in android system.

yunhanw-google, 2025-02-03:
android chip-tool only generate apk target instead of chip-tool arm-32 bit command line binary, this is per designed. thanks

closed 2025-02-03 by yunhanw-google