project-chip/connectedhomeip #36444: [1.3] Can't find `SetRFIDCode` command in chip-tool

lboue, 2024-11-09:
### Reproduction steps

Can't find `SetRFIDCode` command in chip-tool

### Bug prevalence

always

### GitHub hash of the SDK that was being used

0aa26565d8954a2c36da5ffae101c4640263cf41

### Platform

core

### Platform Version(s)

1.3

### Type

Common Cluster Logic

### Anything else?

There is 46 command in the `DoorLock` data_model :
SetRFIDCode
```
~/connectedhomeip# grep 'command id' data_model/1.3/clusters/DoorLock.xml
    <command id="0x00" name="LockDoor" direction="commandToServer" response="Y">
    <command id="0x01" name="UnlockDoor" direction="commandToServer" response="Y">
    <command id="0x02" name="Toggle" direction="commandToServer" response="Y">
    <command id="0x03" name="UnlockWithTimeout" direction="commandToServer" response="Y">
    <command id="0x04" name="GetLogRecord" direction="commandToServer" response="GetLogRecordResponse">
    <command id="0x04" name="GetLogRecordResponse" direction="responseFromServer">
    <command id="0x05" name="SetPINCode" direction="commandToServer" response="Y">
    <command id="0x06" name="GetPINCode" direction="commandToServer" response="GetPINCodeResponse">
    <command id="0x06" name="GetPINCodeResponse" direction="responseFromServer">
    <command id="0x07" name="ClearPINCode" direction="commandToServer" response="Y">
    <command id="0x08" name="ClearAllPINCodes" direction="commandToServer" response="Y">
    <command id="0x09" name="SetUserStatus" direction="commandToServer" response="Y">
    <command id="0x0A" name="GetUserStatus" direction="commandToServer" response="GetUserStatusResponse">
    <command id="0x0A" name="GetUserStatusResponse" direction="responseFromServer">
    <command id="0x0B" name="SetWeekDaySchedule" direction="commandToServer" response="Y">
    <command id="0x0C" name="GetWeekDaySchedule" direction="commandToServer" response="GetWeekDayScheduleResponse">
    <command id="0x0C" name="GetWeekDayScheduleResponse" direction="responseFromServer">
    <command id="0x0D" name="ClearWeekDaySchedule" direction="commandToServer" response="Y">
    <command id="0x0E" name="SetYearDaySchedule" direction="commandToServer" response="Y">
    <command id="0x0F" name="GetYearDaySchedule" direction="commandToServer" response="GetYearDayScheduleResponse">
    <command id="0x0F" name="GetYearDayScheduleResponse" direction="responseFromServer">
    <command id="0x10" name="ClearYearDaySchedule" direction="commandToServer" response="Y">
    <command id="0x11" name="SetHolidaySchedule" direction="commandToServer" response="Y">
    <command id="0x12" name="GetHolidaySchedule" direction="commandToServer" response="GetHolidayScheduleResponse">
    <command id="0x12" name="GetHolidayScheduleResponse" direction="responseFromServer">
    <command id="0x13" name="ClearHolidaySchedule" direction="commandToServer" response="Y">
    <command id="0x14" name="SetUserType" direction="commandToServer" response="Y">
    <command id="0x15" name="GetUserType" direction="commandToServer" response="GetUserTypeResponse">
    <command id="0x15" name="GetUserTypeResponse" direction="responseFromServer">
    <command id="0x16" name="SetRFIDCode" direction="commandToServer" response="Y">
    <command id="0x17" name="GetRFIDCode" direction="commandToServer" response="GetRFIDCodeResponse">
    <command id="0x17" name="GetRFIDCodeResponse" direction="responseFromServer">
    <command id="0x18" name="ClearRFIDCode" direction="commandToServer" response="Y">
    <command id="0x19" name="ClearAllRFIDCodes" direction="commandToServer" response="Y">
    <command id="0x1A" name="SetUser" direction="commandToServer" response="Y">
    <command id="0x1B" name="GetUser" direction="commandToServer" response="GetUserResponse">
    <command id="0x1C" name="GetUserResponse" direction="responseFromServer">
    <command id="0x1D" name="ClearUser" direction="commandToServer" response="Y">
    <command id="0x20" name="OperatingEventNotification" direction="commandToClient">
    <command id="0x21" name="ProgrammingEventNotification" direction="commandToClient">
    <command id="0x22" name="SetCredential" direction="commandToServer" response="SetCredentialResponse">
    <command id="0x23" name="SetCredentialResponse" direction="responseFromServer">
    <command id="0x24" name="GetCredentialStatus" direction="commandToServer" response="GetCredentialStatusResponse">
    <command id="0x25" name="GetCredentialStatusResponse" direction="responseFromServer">
    <command id="0x26" name="ClearCredential" direction="commandToServer" response="Y">
    <command id="0x27" name="UnboltDoor" direction="commandToServer" response="Y">
```

bzbarsky-apple, 2024-11-16:
SetRFIDCode is not part of Matter.  It's Zigbee-only.  Note the `!USR & RID` conformance, and the fact that in the device library the conformance for USR is `Matter & (PIN | RID | FPG | FACE | ALIRO )`.

So as long as `Matter` is true, `!USR & RID` will be false.

closed 2024-11-16 by bzbarsky-apple

bzbarsky-apple, 2024-11-19:
Because the spec is silly and the auto-scraper is not very smart.