project-chip/connectedhomeip #36448: [BUG] Missing Release/Tag v1.4

mmt-10, 2024-11-11:
### Reproduction steps

I heard to release Matter v1.4[https://csa-iot.org/newsroom/matter-1-4-enables-more-capable-smart-homes/](https://csa-iot.org/newsroom/matter-1-4-enables-more-capable-smart-homes/)

However, I could not find Tag v1.4 in SDK.
[https://github.com/project-chip/connectedhomeip/tags](https://github.com/project-chip/connectedhomeip/tags)

### Bug prevalence

always

### GitHub hash of the SDK that was being used

c3b35eb6c16437ad19d919faa69d269c95da10db

### Platform

core

bzbarsky-apple, 2024-11-16:
See https://github.com/project-chip/connectedhomeip/tree/v1.4-branch

closed 2024-11-16 by bzbarsky-apple

dtodor, 2024-11-20:
Is there going to be a tag for the v1.4.0.0 release? What is the correct hash of the release? The v1.4 branch has recent commits.