project-chip/connectedhomeip #36455: [1.4] CHIP:ZCL: Shutting down thermostat server cluster on endpoint 2

lboue, 2024-11-11:
### Reproduction steps

Thermostat server is not starting on EP in Water Heater [Matter Linux Energy Management Example](https://github.com/project-chip/connectedhomeip/tree/master/examples/energy-management-app/linux)
`[1731319971.337995][3627:3627] CHIP:ZCL: Shutting down thermostat server cluster on endpoint 2
`

**Build***
```
    ./scripts/run_in_build_env.sh "./scripts/build/build_examples.py --target linux-x64-energy-management-no-ble-asan-clang build"
    ./out/linux-x64-energy-management-no-ble-asan-clang/chip-energy-management-app
```

### Bug prevalence

always

### GitHub hash of the SDK that was being used

c3b35eb6c16437ad19d919faa69d269c95da10db

### Platform

other

### Type

Common Cluster Logic

### Anything else?

```
[1731319971.337767][3627:3627] CHIP:DL: Device Configuration:
[1731319971.337815][3627:3627] CHIP:DL:   Serial Number: TEST_SN
[1731319971.337857][3627:3627] CHIP:DL:   Vendor Id: 65521 (0xFFF1)
[1731319971.337874][3627:3627] CHIP:DL:   Product Id: 32769 (0x8001)
[1731319971.337879][3627:3627] CHIP:DL:   Product Name: TEST_PRODUCT
[1731319971.337897][3627:3627] CHIP:DL:   Hardware Version: 0
[1731319971.337903][3627:3627] CHIP:DL:   Setup Pin Code (0 for UNKNOWN/ERROR): 20202021
[1731319971.337906][3627:3627] CHIP:DL:   Setup Discriminator (0xFFFF for UNKNOWN/ERROR): 3840 (0xF00)
[1731319971.337923][3627:3627] CHIP:DL:   Manufacturing Date: (not set)
[1731319971.337927][3627:3627] CHIP:DL:   Device Type: 1292 (0x50C)
[1731319971.337947][3627:3627] CHIP:SVR: SetupQRCode: [MT:-24J0AFN00KA0648G00]
[1731319971.337953][3627:3627] CHIP:SVR: Copy/paste the below URL in a browser to see the QR Code:
[1731319971.337956][3627:3627] CHIP:SVR: https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT%3A-24J0AFN00KA0648G00
[1731319971.337965][3627:3627] CHIP:SVR: Manual pairing code: [34970112332]
[1731319971.337979][3627:3627] CHIP:SVR: Energy Management App: ApplicationInit()
[1731319971.337995][3627:3627] CHIP:ZCL: Shutting down thermostat server cluster on endpoint 2
[1731319971.338006][3627:3627] CHIP:DMG: Endpoint 0, Cluster 0x0000_001D update version to 704c4173
[1731319971.338011][3627:3627] CHIP:DMG: Endpoint 0, Cluster 0x0000_001D update version to 704c4174
[1731319971.338047][3627:3627] CHIP:SVR: EVSE: Initializing EvseTargetsDelegate
[1731319971.338086][3627:3627] CHIP:SVR: EVSE: Unable to restore persisted ChargingEnabledUntil value
[1731319971.338108][3627:3627] CHIP:SVR: EVSE: Unable to restore persisted DischargingEnabledUntil value
[1731319971.338129][3627:3627] CHIP:SVR: EVSE: Unable to restore persisted UserMaximumChargeCurrent value
[1731319971.338163][3627:3627] CHIP:SVR: EVSE: Unable to restore persisted RandomizationDelayWindow value
[1731319971.338183][3627:3627] CHIP:SVR: EVSE: Unable to restore persisted ApproximateEVEfficiency value
[1731319971.338221][3627:3627] CHIP:SVR: mPowerMode updated to 2
[1731319971.338226][3627:3627] CHIP:DMG: Endpoint 1, Cluster 0x0000_0090 update version to 40285b13
[1731319971.338237][3627:3627] CHIP:DMG: Endpoint 0, Cluster 0x0000_002F update version to 1e15a823
[1731319971.338243][3627:3627] CHIP:DMG: Endpoint 0, Cluster 0x0000_002F update version to 1e15a824
[1731319971.338248][3627:3627] CHIP:DMG: Endpoint 0, Cluster 0x0000_002F update version to 1e15a825
[1731319971.338261][3627:3627] CHIP:DMG: Endpoint 0, Cluster 0x0000_002F update version to 1e15a826
[1731319971.338268][3627:3627] CHIP:DMG: Endpoint 0, Cluster 0x0000_002F update version to 1e15a827
[1731319971.338275][3627:3627] CHIP:SVR: mEsaState updated to 1
[1731319971.338287][3627:3627] CHIP:DMG: Endpoint 1, Cluster 0x0000_0098 update version to 7e1b57cb
[1731319971.338291][3627:3627] CHIP:SVR: mAbsMinPower updated to 0000000000124F80
[1731319971.338294][3627:3627] CHIP:DMG: Endpoint 1, Cluster 0x0000_0098 update version to 7e1b57cc
[1731319971.338298][3627:3627] CHIP:SVR: mAbsMaxPower updated to 000000000073F780
[1731319971.338301][3627:3627] CHIP:DMG: Endpoint 1, Cluster 0x0000_0098 update version to 7e1b57cd
[1731319971.347862][3627:3627] CHIP:DIS: Updating services using commissioning mode 1
[1731319971.354361][3627:3627] CHIP:DIS: CHIP minimal mDNS started advertising.
```

bzbarsky-apple, 2024-11-16:
@jamesharrow

jamesharrow, 2025-05-16:
At present the thermostat cluster is not enabled in the water heater example app. It is known that this example fails the spec device type conformance tests, however the original focus of this example was to allow testing of the water heater management cluster.

This remains a development task on the backlog.
It may just be a case of enabling it in Zap however due to the way the example app can be either EVSE or Water heater selected at runtime on Linux this makes it more challenging (EP1 is EVSE and EP2 is water heater) so it may require a dynamic endpoint code to activate it.

Other platforms (like ESP32) have this as a build time configuration.

Which platform are you building for?

lboue, 2025-05-16:
> Which platform are you building for?

Thanks for the explanation. I tried with Linux and Silabs.