project-chip/connectedhomeip #36480: [BUG] Use MatterSupport to add thread sensor, will popup "No thread border router" in app

JoeyYang2010, 2024-11-13:
### Reproduction steps

[image]
Here is my sample code:
    func testMatter() async  {
        let homes = [MatterAddDeviceRequest.Home(displayName: "My Home")]
        let topology = MatterAddDeviceRequest.Topology(ecosystemName: "MyEcosystemName", homes: homes)
        let request2 = MatterAddDeviceRequest(topology: topology)
        Task
        {
            do
            {
//                try await request2.perform()
//                let userDefaults = UserDefaults(suiteName: "group.sercomm.SampleHybridApp1")
//                retrievedPayload = userDefaults!.string(forKey: "onboardingPayload") ?? ""

Do I have a way to not show apple's picker and not to add device to apple home fabric first,I just want to add to my own fabric.

### Bug prevalence

Each time

### GitHub hash of the SDK that was being used

2024/4/19 master version

### Platform

darwin

JoeyYang2010, 2024-11-13:
@bzbarsky-apple Can you help check this issue.

bzbarsky-apple, 2024-11-15:
@JoeyYang2010 Please follow the instructions at https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#providing-feedback-to-apple and put the ticket ID here.  This does not look like a Matter SDK issue.

closed 2024-11-15 by bzbarsky-apple

JoeyYang2010, 2024-11-22:
@bzbarsky-apple 
    override func selectThreadNetwork(from threadScanResults: [MatterAddDeviceExtensionRequestHandler.ThreadScanResult]) async throws -> MatterAddDeviceExtensionRequestHandler.ThreadNetworkAssociation {
        // Use this function to select a Thread network for the device if your ecosystem has special requirements.
        // Or, return `.defaultSystemNetwork` to use the default Thread network.
                return .defaultSystemNetwork
    }

bzbarsky-apple, 2024-11-22:
@JoeyYang2010 I have no idea.  Please see above for the right link for Apple-specific things.