project-chip/connectedhomeip #36505: [CI] Find a way to run CNET tests in CI

cecille, 2024-11-14:
### Reproduction steps / Feature

The linux overrides on the network commissioning cluster are (almost? entirely?) non-spec-compliant since they run-time swap ethernet / wifi / thread and do not change the statically generated zap attributes for this cluster. In order to run these tests in CI we need
- properly run-time configurable cluster
- a way to simulate network connections / swaps scans etc. in the CI, either through something like cirque, or though test mocks

### Platform

core (please add to version below)

### Type

Manually tested with SDK

r-recchia, 2025-08-21:
Problems with flask, it seems that it is not installed correctly. This is strange because flask is used for cirque tests

FrancoLionti, 2026-09-08:
**Status update (2026-09-08)**

---

## TL;DR

- **Issue goal (run CNET tests in CI without real Wi‑Fi / Thread hardware):** still **open**.
- **Cluster attribute lists matching the active network type:** **mostly done**: the code-driven cluster updates them at runtime.
- **CNET tests in CI today:** **2 of 13**: only `TC_CNET_4_3` and `TC_CNET_1_4` (Ethernet).
- **`--wifi` / `--thread` in CI:** **not enough**: real Linux drivers, empty network list without a radio, Thread often crashes the app in GitHub Actions.
- **Simulated network driver in `all-clusters-app`:** **not built yet**: test-only code exists in the repo but is not wired to the example app.

---

## What still needs to be built

1. **Simulated network driver** for `all-clusters-app`: pretend there is a connected network, return fixed scan results, answer connect/remove commands sensibly, support failsafe revert. No real radio, and no change to how the CI machine itself is networked.

2. **Separate CI test runs**: Wi‑Fi / Thread CNET tests must not share the same app run as the Ethernet tests, or `TC_CNET_4_3` skips and the job fails (`--fail-on-skipped`).

3. **Cirque (or similar) for a few tests only**: `4_11`, `4_12`, `4_23` need the host to actually switch networks. That is on top of the simulated driver, not a substitute for it.

When this issue was opened, one worry was that static ZAP config did not match what the Linux app exposes at runtime. The code-driven cluster fixed most of that. What is still missing is the **network driver** behind the cluster and **CI setup** to use it.

---

## What works today

**On `master` in the normal Python CI job:**

- `TC_CNET_4_3`: Ethernet attributes
- `TC_CNET_1_4`: runs, but only the Ethernet path (secondary-interface steps
  are skipped)

**Throwaway draft PR [#73880](https://github.com/project-chip/connectedhomeip/pull/73880):**

- Second `all-clusters-app` build with Wi‑Fi enabled → **worked**
- `TC_CNET_4_15` with `--wifi`, no radio → **passed** (~11 s; test only checks “unknown network ID → not found”)
- Existing Ethernet tests on the default app → **still passed**
- `TC_CNET_4_16` with `--thread` only → **failed** (app crashes on startup; no Thread D-Bus in the CI container)
- `TC_CNET_1_4` with `--thread` and `--wifi` → **failed** (same startup crash)

Failed Thread run logs:
[GHA job 33644410875](https://github.com/project-chip/connectedhomeip/actions/runs/33644410875)

---

## What does not work today

- **`--wifi` on the app CI already builds**: Wi‑Fi is compiled out of that build, so the flag does nothing.
- **`--thread` on the default CI app**: startup needs Thread over D-Bus; the service is missing, so the app exits before the test runs.
- **Real Wi‑Fi / Thread drivers without hardware**: `Networks` stays empty; tests that need one connected network fail.
- **`--wifi` in the same CI run as Ethernet tests**: Ethernet test skips → job fails.
- **Unit-test `FakeWiFiDriver` as-is**: not hooked to the example app; does not model a connected network properly.
- **Cirque + Flask (Aug 2025 comments)**: blocked on Flask; no linked PR yet.
- **`TC_CNET_4_9` / `4_10` with on-network commissioning**: test runner does not pass `--wifi-ssid` / `--thread-dataset-hex` for that mode.

---

## Main blockers

- No `--simulated-wifi` / `--simulated-thread` (or equivalent) in Linux example apps
- Thread startup is fatal in CI when OpenThread’s D-Bus service is absent
- Test runner does not apply Wi‑Fi / Thread PIXIT values for on-network commissioning
- CI PICS file is Ethernet-only: Wi‑Fi / Thread runs need their own PICS or must omit `--PICS`
- Cirque workflow does not run on pull requests, so Cirque tests would not gate merges

---

## Suggested next steps (small PRs)

| Step | What                                                                 | Tests helped                        |
| ---- | -------------------------------------------------------------------- | ----------------------------------- |
| 0    | Fix test runner for on-network Wi‑Fi SSID / Thread dataset           | `4_9`, `4_10`                       |
| A    | Simulated Wi‑Fi driver + Wi‑Fi CI build + `4_1` / `4_15`             | Wi‑Fi attributes, network-not-found |
| B    | Simulated Thread driver (no real OpenThread) + `4_2`, `4_16`, `4_22` | Thread in CI without ot-agent       |
| C    | Both simulated drivers + `TC_CNET_1_4` run2                          | Secondary endpoint `0xFFFE`         |
| D    | Enable `4_4`, `4_9`, `4_10`                                          | Connect / remove / failsafe         |
| E    | Cirque for `4_11`, `4_12`, `4_23`                                    | Real network switch                 |

The simulated driver should answer correctly (e.g. “network not found” for a bad ID), not return success for everything, and not reconfigure the CI runner’s network.

---

## Still open

- **PR #73880**: proved CI wiring only; next step is a real simulated driver, not `--wifi` on an empty network list.
- **Cirque / Flask**: possible Docker / Flask issue in `chip-build-cirque`; not confirmed.
- **`TC_CNET_4_11` / `4_23`**: missing from `test_metadata.yaml` “not automated” list.

---

## Links

- Draft experiment: https://github.com/project-chip/connectedhomeip/pull/73880
- Failed `--thread` CI run: https://github.com/project-chip/connectedhomeip/actions/runs/33644410875