project-chip/connectedhomeip #36511: [Feature]  In the esp32 platform, sometimes disconnection occurs when joining a family. Is there an API status value in the code about the reason why the device is disconnected? And an explanation of the specific reasons for the offline status value.

wolen666, 2024-11-15:
### Feature description

...

In the esp32 platform, sometimes disconnection occurs when joining a family. Is there an API status value in the code about the reason why the device is disconnected? And an explanation of the specific reasons for the offline status value.

### Platform

esp32

### Platform Version(s)

1.2

bzbarsky-apple, 2024-11-16:
@dhrishi @shubhamdp

dhrishi, 2025-02-22:
@wolen666 Sorry for missing this.

>sometimes disconnection occurs when joining a family.

Which interface are you talking about here? Is it while connecting to Wi-Fi or BLE or anything else?

wolen666, 2025-02-24:
[image]

Similar to the picture above, when HomeKit shows that it is joining the family, I need to add the corresponding Freertos task to the device at this moment. Is there an API that can mark it as "joining the family"?