project-chip/connectedhomeip #36512: [Workflow] Cirque CI is failing

lboue, 2024-11-15:
### Workflow issues

I have multiple Cirque CI jobs failing:

- #36429
  - https://github.com/project-chip/connectedhomeip/actions/runs/11853007697/job/33032275621?pr=36429
- #36441
  - Cirque CI job: https://github.com/project-chip/connectedhomeip/actions/runs/11853006285/job/33032268252?pr=36441

### Platform

core (please add to version below)

### Anything else?

```
Run integrations/docker/images/stage-2/chip-build-cirque/run.sh \
81: Pulling from project-chip/chip-build-cirque
Digest: sha256:b15[24](https://github.com/project-chip/connectedhomeip/actions/runs/11853006285/job/33032268252?pr=36441#step:12:25)c86ef202d5ec9e708546e1bc264c8ee691865596410892b80cbbfe0b76e
Status: Image is up to date for ghcr.io/project-chip/chip-build-cirque:81
ghcr.io/project-chip/chip-build-cirque:81
Logs will be stored at /tmp/cirque_test_output
[ RUN] EchoTest
[PASS] EchoTest
[ RUN] EchoOverTcpTest
[PASS] EchoOverTcpTest
[ RUN] FailsafeTest
[FAIL] FailsafeTest (Exitcode: 1)
[ RUN] MobileDeviceTest
[FAIL] MobileDeviceTest (Exitcode: 1)
[ RUN] CommissioningTest
[FAIL] CommissioningTest (Exitcode: 1)
[ RUN] IcdWaitForActiveTest
[FAIL] IcdWaitForActiveTest (Exitcode: 1)
[ RUN] SplitCommissioningTest
[FAIL] SplitCommissioningTest (Exitcode: 1)
[ RUN] CommissioningFailureTest
[FAIL] CommissioningFailureTest (Exitcode: 1)
[ RUN] CommissioningFailureOnReportTest
[FAIL] CommissioningFailureOnReportTest (Exitcode: 1)
[ RUN] PythonCommissioningTest
[FAIL] PythonCommissioningTest (Exitcode: 1)
[ RUN] CommissioningWindowTest
[FAIL] CommissioningWindowTest (Exitcode: 1)
[ RUN] SubscriptionResumptionTest
[FAIL] SubscriptionResumptionTest (Exitcode: 1)
[ RUN] SubscriptionResumptionCapacityTest
[FAIL] SubscriptionResumptionCapacityTest (Exitcode: 1)
[ RUN] SubscriptionResumptionTimeoutTest
[FAIL] SubscriptionResumptionTimeoutTest (Exitcode: 1)
[INFO] Logs will be uploaded to artifacts.
[FAIL] Test failed, test log can be found at /tmp/cirque_test_output
Error: Process completed with exit code 1.
```

bzbarsky-apple, 2024-11-16:
@yunhanw-google

yunhanw-google, 2024-11-16:
we need cherry-pick this https://github.com/project-chip/connectedhomeip/pull/36407 to 1.4 branch

yunhanw-google, 2024-11-16:
https://github.com/project-chip/connectedhomeip/pull/36523

lboue, 2024-11-18:
Fixed by:
- https://github.com/project-chip/connectedhomeip/pull/36523

closed 2024-11-18 by lboue