project-chip/connectedhomeip #36534: [Test Failed] [TC-CNET-4.3] Step 7: NetworkID value is empty

FIREWORKMBY, 2024-11-18:
### Test issue(s)

[](url)The error step command is：./chip-tool networkcommissioning read networks 1 0
The expected result is：
[1654076774.294361][33659:33664] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Attribute 0x0000_0001 DataVersion: 609053543
[1654076774.294505][33659:33664] CHIP:TOO:   Networks: 1 entries
[1654076774.294625][33659:33664] CHIP:TOO:     [1]: {
[1654076774.294679][33659:33664] CHIP:TOO:       NetworkID: 6368697035
[1654076774.294726][33659:33664] CHIP:TOO:       Connected: TRUE
[1654076774.294774][33659:33664] CHIP:TOO:      }

The error result is：
[1730188060.918] [73112:73120] [TOO] Endpoint: 0 Cluster: 0x0000_0031 Attribute 0x0000_0001 DataVersion: 3678178638
[1730188060.918] [73112:73120] [TOO]   Networks: 1 entries
[1730188060.918] [73112:73120] [TOO]     [1]: {
[1730188060.918] [73112:73120] [TOO]       NetworkID:
[1730188060.918] [73112:73120] [TOO]       Connected: TRUE
[1730188060.918] [73112:73120] [TOO]      }

The NetworkID parameter is empty.Do we need to do more to read NetworkID?
More logging reports for this use case are:
[TC-CNET-4.3—Manual-test-error.log.txt](https://github.com/user-attachments/files/17798896/TC-CNET-4.3.Manual-test-error.log.txt)

### Platform

raspi

### Anything else?

[](url)

bzbarsky-apple, 2024-11-18:
@FIREWORKMBY Sounds like an issue with the server.  What exactly is the server running?  Is the server a raspi as listed under "Platform" or something else?

FIREWORKMBY, 2024-11-19:
@bzbarsky-apple 
Equipment used for testing:
Test hraness: Raspberry PI
DUT server: An openwrt device, which is an intelligent gateway with an OpenThread network inside,the other network interfaces are as follows
[image]

bzbarsky-apple, 2024-11-19:
> DUT server: An openwrt device

OK, sounds like an issue with that device, then.  Which exact "platform" of Matter is running on that?  Linux?  Something else?

FIREWORKMBY, 2024-11-20:
> > DUT server: An openwrt device
> 
> OK, sounds like an issue with that device, then. Which exact "platform" of Matter is running on that? Linux? Something else?

Yes, the linux routine /examples/network-manager-app , and then ported to openwrt platform  called Hoomax

bzbarsky-apple, 2024-11-20:
Well, it sounds like its platform-specific bits are not returning a proper network id.  Debug and fix those?